mu-editor/mu #148: Mu Mac OS signed builds

gbaman, 2016-09-19:
Right now, Gatekeeper on Mac OS will refuse to run Mu if it is set to default on 10.8 and beyond as it is unsigned. With Mac OS 10.12 Sierra, it is looking like the ability to globally ignore these errors is going away (from GUI anyway using "Anywhere"). It will still be possible to install unsigned apps with right click -> open as you can do now, but that is a pain and may also need to be documented in the mean time as is a bit obscure.

carlosperate, 2016-09-19:
Yes, being able to sign the application to avoid the warning and "workaround" would be great. I don't really have any experience on that area, but I was under the impression that we would need a (paid) Apple App Developer account. So any help on that area would be quite welcomed.

ntoll, 2016-09-22:
I can get in touch with the PSF - it's exactly the sort of thing they pay for. I'll post updates here. In the first instance I'll make a grant request for both OSX and Windows since I believe Window works in the same sort of way (signed installation packages are much better).

ntoll, 2018-06-04:
Closing since it's coming very soon with the 1.0 release.

closed 2018-06-04 by ntoll

ukBaz, 2018-07-16:
I have just downloaded mu-editor_rc1_osx.dmg and get the following pop up.

[image]

Is that expected?

carlosperate, 2018-07-16:
Hi Barry,
Thanks for checking. Yes, this is expected at the moment as the signing hasn't been done yet. I believe @ntoll's plan is to start signing the app when it reaches the  v1.0.0 release.
https://codewith.mu/en/howto/install_macos

ntoll, 2018-07-16:
Hi @ukBaz, @carlosperate,

Literally just minutes ago, I *think* I got the correct certificates for signing the app. Sadly, since this isn't an app developed within Apple's ecosystem, it's a bit of a dark art to know what's supposed to happen. However, I think I have everything I need in terms of the cryptography and the tools needed to sign the .app for Mu. All I need to work out is how to put them together... and I've never done this before. That'll likely happen on Thursday, given how my week is panning out.

I'll ping you both via this issue to point you at the (hopefully correctly) signed app for testing.

(I'm also doing this for Windows too, although this is even more of a pain in the proverbial since I've had to pay a not unsubstantial amount of money for the bureaucracy and even a notary to meet me face to face to prove I am who I say I am. Sadly, this process is geared towards companies rather than independent developers. Hey ho... it'll be over soon).

N.