mu-editor/mu #150: Markdown the website

dannystaple, 2016-09-19:
So it is easier to contribute.
Consider jekyll bootstrap.

ZanderBrown, 2016-09-19:
Definitely a good idea. particularly when we start writing the help pages

carlosperate, 2016-09-26:
This has been implemented in https://github.com/mu-editor/mu-editor.github.io/pull/3, although we might still need to discuss how the help documentation creation works (https://github.com/mu-editor/mu-editor.github.io/issues/4).

ntoll, 2016-10-23:
I just copied it over. :-)

It might be better to simply move the help docs over from their own repos.

Not sure why they were separated out in the first place..?!?!? Anyone?

carlosperate, 2016-10-23:
> Not sure why they were separated out in the first place..?!?!? Anyone?

Why the help files were separated?
A repository like https://github.com/mu-editor/mu-editor.github.io can be quite intimidating to edit if you don't know what's going on (and I must admit I still struggle!). Having a repository only with the markdown files is a lot more welcoming.
Then the idea was to have a git submodule/s inside https://github.com/mu-editor/mu-editor.github.io, which is what https://github.com/mu-editor/mu-editor.github.io/pull/8 was aiming to do.

carlosperate, 2017-09-27:
Already implemented in https://github.com/mu-editor/mu-editor.github.io/pull/9, the markdown help pages can be edited in https://github.com/mu-editor/help-pages

closed 2017-09-27 by carlosperate