mu-editor/mu #151: In the website move the downloads to releases

dannystaple, 2016-09-19:
Move executables and tarballs to github releases  - keep the clones small.
Ensure any website links to download will still reach the releases before merging.

ntoll, 2016-09-22:
Any docs on how to do this?

carlosperate, 2016-09-22:
https://help.github.com/articles/creating-releases/

It's basically a git tag with additional content attached to it in GitHub. An advantage is to be able to retrieve all this data from the GitHub API: https://developer.github.com/v3/repos/releases/

ntoll, 2016-09-22:
OK... I'll take a look.

carlosperate, 2016-10-15:
There is a related discussion to this one in https://github.com/mu-editor/mu/issues/155 (linking the issues to easily keep track).

ntoll, 2016-10-23:
Fixed. See the brand new release associated with this repos!

closed 2016-10-23 by ntoll