mu-editor/mu #154: Click on the current REPL line should place cursor at the end

carlosperate, 2016-09-19:
Currently there is some unexpected behaviour if REPL active prompt line (the last line where you type the code you'd like to execute) is edited using the mouse.

So to reproduce we can start by having some unexecuted text at the active prompt line:

```
>>> MicroPython v1.7-9-gbe020eb on 2016-04-18; micro:bit with nRF51822
Type "help()" for more information.
>>> display.scroll("test")
```

Then use the mouse to place the cursor position somewhere other than the end of the line. In this case we can place it at the open quotation marks.

```
>>> display.scroll(" test")
                    ^ Click there
```

Typing some text at this point will overwrite the displayed text. So by typing `1234` we end up with:

```
>>> display.scroll("1234")
```

If we try to execute this line of code it will result in a syntax error, because the code actually sent to the microbit is: 

````
display.scroll("test")1234
````

I saw this problem a few time during the PyCon UK kids day workshops and caused some obvious confusion.

ntoll, 2016-09-22:
Good catch. I guess we should not let people move the cursor with the mouse (a UI layer issue). Also, given that we want to remove the QT based dependencies of the editor I guess this is something we should address in a PySerial version of the REPL.

carlosperate, 2016-09-22:
It would still be useful to be able to use the mouse on the REPL to copy the contents, which is why my first though was to programmatically move the cursor to the end of the line. If we can stop the cursor being change on mouse click, while still maintaining text selection available, that would probably be the best solution.
As far as the move to PySerial goes, we will probably still be dependant of the QTextEdit widget, so this can be address independently.

ZanderBrown, 2016-09-22:
@carlosperate true Qt will still be required for widgets but the QSerial bit is part of a separate package

carlosperate, 2016-09-22:
No yeah, of course, but QSerialPort or PySerial only deal with the data in and out the port. How it is displayed and where the cursor is placed would still be controlled by the QTextEdit widget, which is what I meant by saying that this is something to be worked on that area rather than the Serial Port.

ZanderBrown, 2016-09-22:
Ah right sorry. Still, it's a good excuse to complete the transition

askvictor, 2016-10-10:
Might be nice to (if the user has clicked in the 'editable' part of the command), to emulate some arrow keys to send the device's cursor to where the UI cursor thinks it should be.

@ntoll - what is the plan around removing QT dependencies? Is there a roadmap or discussion group? I ask in part as my esp8266/generic-serial-upython-device fork has gone the other way and has abstracted the device entirely, using the Qt serial library for repl and fs, while using the Qt websockets lib for connecting via webrepl.

ntoll, 2016-10-11:
Good morning!

First of all, many apologies for my relative absence from Mu - "real work" is very busy at the moment (Christmas has arrived - literally - http://christmasfood.marksandspencer.com/). However, it's wonderful to see all the pending contributions. 

I'll be at PyCon Poland from Thursday -> Sunday and hope to find some hours to catch up on Mu things.

Regarding QT dependencies - the plan is to remove/move QT dependent functions (such as the REPL) and have them encapsulated solely in the logic layer with obvious ways in which to plug UI thing into this layer. It should be relatively simple to build a UI from and GUI toolkit on top of this foundation. I'll make a spike, try to write this up over the PyCon PL weekend and I'd love some feedback.

Thanks for your patience. :-)

ntoll, 2018-06-04:
OK,... I'm closing this because:

* You should be able to move the blinky cursor anywhere in the REPL since this is needed to be able to drag and select text.
* If you place the cursor in a not-at-the-end-of-the-line position then type, the cursor (and your typed text) appears at the end of the line. This is IMHO the correct behaviour.
* The second point should make it clear to users to use the arrow keys instead when "editing" a new line with the cursor.

Thanks for the contributions.

carlosperate, 2018-06-04:
> You should be able to move the blinky cursor anywhere in the REPL since this is needed to be able to drag and select text.

Good point, agreed.

> If you place the cursor in a not-at-the-end-of-the-line position then type, the cursor (and your typed text) appears at the end of the line. This is IMHO the correct behaviour.
> The second point should make it clear to users to use the arrow keys instead when "editing" a new line with the cursor.

Completely agree, but this is not the current behaviour:

[image]

ntoll, 2018-06-04:
Ahh... ok. I see what you mean. You're right to re-open. 

Hmmm... not sure what to suggest as a way of implementing a solution. I guess some potentially complicated "if the click is on the line that's currently pending" edge case that sets a flag. 

Thoughts..?

ntoll, 2018-06-05:
Hmm... this requires some thought in order to find the simplest solution. Personally, I think this should be a post-1.0 fix and we can let it mull over in our heads in the meantime.

carlosperate, 2018-06-05:
Agreed

carlosperate, 2018-06-05:
Hope you don't mind, I've created a couple of GH milestones to track issue priorities.

ntoll, 2018-06-05:
+1 on milestones. :-)

dybber, 2020-09-27:
This should be fixed when #1026 is merged.

closed 2020-09-29 by dybber