mu-editor/mu #161: Installing on Linux - missing dependencies/libraries

catchpolej, 2016-09-23:
Ubuntu 12.04.5 LTS

RUNNING: sudo apt-get install mu
Messages are..... 
Reading package lists... Done
Building dependency tree       
Reading state information... Done
mu is already the newest version.
You might want to run 'apt-get -f install' to correct these:
The following packages have unmet dependencies.
 mu : Depends: python3:any (>= 3.3.2-2~)
      Depends: python3-setuptools but it is not going to be installed
      Depends: python3-pyqt5 but it is not installable
      Depends: python3-pyqt5.qsci but it is not installable
      Depends: python3-pyqt5.qtserialport but it is not installable
      Depends: python3-serial but it is not going to be installed
      Depends: python3-pep8 but it is not installable
      Depends: pyflakes but it is not going to be installed
E: Unmet dependencies. Try 'apt-get -f install' with no packages (or specify a solution).

after running apt-get -f install and retrying:
Package mu is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

E: Package 'mu' has no installation candidate

RUNNING  mu-2016-09-20_15_27_05 installer:
Error loading Python lib '/tmp/_MEIL5bsvx/libpython3.4m.so.1.0': /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.17' not found (required by /tmp/_MEIL5bsvx/libpython3.4m.so.1.0)

Python 3.4 is installed.
Any ideas? Thanks

carlosperate, 2016-09-23:
I'm assuming you have already tried to install the Mu deb package from http://ardublockly-builds.s3-website-us-west-2.amazonaws.com/index.html?prefix=microbit/raspberry_pi/ as otherwise
`sudo apt-get install mu` should probably fail to find it.

Mu needs the following packages:

```
python3
python3-setuptools
python3-pyqt5
python3-pyqt5.qsci
python3-pyqt5.qtserialport
python3-serial
python3-pep8
pyflakes
```

If Python3 with setuptools is already installed, by using pip3 for `pyserial`, `pep8`/`pycodestyle` and `pyflakes` we could reduce it to only:

```
python3-pyqt5
python3-pyqt5.qsci
python3-pyqt5.qtserialport
```

Unfortunately, based on the Ubuntu packages search page (http://packages.ubuntu.com/precise/allpackages), it looks like the PyQt5 packages are not available for 12.04, nor the 12.04 updates or backports repositories.

The `mu-2016-09-20_15_27_05` has been built in 14.04 linking against a newer version of glibc that it cannot find in 12.04. I'm not quite sure if you'll be able to update to the required version, maybe try http://askubuntu.com/questions/736208/upgrading-glibc

I know this was probably not your decision, but is there any reason you are sticking with 12.04? There has been two LTS releases since and 12.04 won't be supported any more in about 6 months.

catchpolej, 2016-09-26:
I have upgraded to Ubuntu 14 over the weekend and mu is now working. Reading and writing to files on the microbit using python is also working (which was my original issue).

Thanks for your help.

Next stage is to get the file data off the microbit onto a computer..........is this documented anywhere?

carlosperate, 2016-09-26:
Have you been able to successfully retrieve the file using mu "files" functionality?

catchpolej, 2016-09-27:
The mu "files" tab comes back with cannot find the microbit. it suggests that the device must have MicroPython flashed onto it before the file system will work? Is micropython not already on it if it can run python programs?

ntoll, 2016-09-27:
Flash the device with an empty file. The device doesn't come with Python "installed" by default.

carlosperate, 2016-09-27:
Yes, if the last thing you've run was a python program from Mu then it should be there. Sometimes after flashing you might have to disconnect it and connect it again. It shouldn't happen often, but when that happens it should show you the same issue if trying to use the REPL.

catchpolej, 2016-09-27:
What is REPL? (replay?) Selecting REPL gives me: Cannot connect to device on port /dev/ttyACM0.
I have flashed with emtpy file from Mu then reflashed with my simple file writer but still same error from Mu "Files" option.

ntoll, 2016-09-27:
Are you in the "dialout" group? For example, type "groups <yourusername>"

If you don't see "dialout" in the resulting list, add yourself to the group like this: sudo useradd -G dialout <yourusername>

catchpolej, 2016-09-27:
I am current working as su as some of the updates I had been doing have had permissions errors even using sudo.
I do not think I have a dialout group?

ntoll, 2016-09-27:
Ubuntu definitely has a dialout group (I use Ubuntu). See updated comment from earlier - GitHub munged my example.

catchpolej, 2016-10-06:
mu files icon/tab still giving me cannot find the microbit but I have been using the ufs functions from the command line instead (https://microbit-micropython.readthedocs.io/en/latest/tutorials/storage.html). Thanks for help.

closed 2016-10-06 by catchpolej

ntoll, 2016-10-06:
That's odd. Mu uses ufs! If ufs works, so should Mu. :-/

Glad to hear you've found a solution.

yurikleb, 2018-02-06:
Hey, I'm having the same REPL problem.
Installed on Ubuntu 16.04, opened MU, it seems like it found the board, and managed to load the code stored on it.
But when I try to open the REPL I get the : "Cannot connect to device on port /dev/ttyACM0" ERROR.
My user is in the "dialout" group.
In Windows everything works fine.

What else could be the issue?

carlosperate, 2018-02-06:
Are you running Mu with as the same user with the same privileges (no sudo?) than the user you have in the `dialout` group?

This sounds like a different issue than the original problem, so to make sure we don't accidentally miss this one, could you open a new ticket? This would also make it easier for other users to find.

luigipacheco, 2018-02-17:
same issue as yurikleb in ubuntu 17.10

yurikleb, 2018-03-01:
Hey @carlosperate sorry for the delay, was away for a few weeks. will open a new ticket now.