mu-editor/mu #163: Change styling for matched braces

carlosperate, 2016-10-01:
It's quite a useful feature, when the cursor is next to a brace and it is highlighted with the matching counterpart.

At the moment is a feature that's only noticeable in the Night theme, when the background of the braces changes to white (which seems to be the [default behaviour](http://pyqt.sourceforge.net/Docs/QScintilla2/classQsciScintilla.html#abf85680f914ee631aa3a513ba823271f)). So we it should be set to a style visible in both themes.

The big question would probably be what type of style. Notepad++ sets a transparent box with a light border around the braces, sublime underlines them, atom underlines on a bright colour, Intellij changes both the background and font colour, etc; so there doesn't seem to be any common standard for this feature. Any ideas?

jezdean, 2016-10-07:
Awesome idea. I can't really say which would be best from an educational perspective. I'd perhaps even say any. The solid dark borders around matching braces is also nice.

Whatever is decided, something different to the style in #164 would be important.

carlosperate, 2016-10-07:
I'll see if this weekend I can prepare a couple of gifs showing different effects, it should make the decision a bit easier.

carlosperate, 2016-10-09:
I found some time today to make this:

[image]

[image]

[image]

[image]

ZanderBrown, 2016-10-09:
I like the look of the last one.

carlosperate, 2016-10-10:
One possible concern is how distracting this highlight could be when using the editor. The box and underline, which are more subtle that the rest, don't really look at that good with the brackets, as it seems to cut through the lower portion of the character.

I personally like the font colour change a bit more than the last one, as it doesn't get too much in the way, yet it is still visible if you look for it. The last one does make it a more clear, so I guess it depends what the general goal of the feature might be (to instruct? to subtly aid? something else?).

ZanderBrown, 2016-10-11:
Very true. the font changing is good too but I think it might be a little non obtrusive and could be unnoticeable

jezdean, 2016-10-11:
I really like the final one where it also highlights the matching parenthesis. You don't _really_ need the colour change as you can see clearly what the problem is without it.

Really awesome.

carlosperate, 2016-10-11:
One thing to keep in mind is that if #164 is implemented it will most likely use the grey square background, so we need to be conscious about having the same visual feedback for a similar (but not the same) feature.

ZanderBrown, 2016-10-12:
I think grey square is the clearest way to do this and then this and #164 + any others will have similar feedback

ntoll, 2018-06-04:
This has been fixed/updated what feels like millions of years ago. Closing... ;-)

closed 2018-06-04 by ntoll