mu-editor/mu #168: Raspbian package out of date

gbaman, 2016-10-09:
Right now, the package available with Raspbian from the Raspbian repos of Mu is rather out of date, [March 2016](http://archive.raspberrypi.org/debian/pool/main/m/mu/). Given the Linux download from the main site doesn't seem to work on Raspbian, any chance we could get an updated copy of Mu in the repos? The March one is missing lots of new cool stuff like radio module support etc.
Apparently the current one is signed by @ntoll.

ntoll, 2016-10-09:
Hey @asb @bennuttall any idea how to update this? If you give me a couple of days I'll update with all the recent PRs and upstream MicroPython fixes.

gbaman, 2016-10-09:
Probably best to throw in @XECDesign as well, given his area of expertise as well.

XECDesign, 2016-10-09:
Happy to accept an updated package whenever it's ready. (serge@<obviousdomain>.org)

asb, 2016-10-21:
It should be enough to just do `git-buildpackage --git-upstream-tree=master -uc -us` on a Debian system to generate an unsigned package based on current master. Sorry, I'm a bit swamped at the moment to help out more.

jg00dman, 2017-02-07:
for raspbian please consider my pull request... fixes issue "Mu interface on small screens #122"

ZanderBrown, 2017-02-07:
@jg00dman please don't add comments to other issues / PRs about a PR you just opened.

We are notified (via email and on GitHub) when you open a PR and it will be reviewed and discussed on it's individual issue

jg00dman, 2017-02-07:
Sorry, it just seemed relevant... I was trying to find something else I could help with.

ntoll, 2018-06-04:
People at RPi are handling this as/when we create new versions of Mu.

closed 2018-06-04 by ntoll