mu-editor/mu #169: Copy/Paste via keys not working in REPL

eduvik, 2016-10-09:
Works fine using context menu, but hotkeys don't work (at least not on Mac).

eduvik, 2016-10-09:
Actually; on a Windows/Linux, Ctrl-C should probably not be mapped to Copy; however on Mac Command-C should be (currently doesn't work). Perhaps follow the conventions of plenty of terminal programs and use Ctrl-Shift-C for copy when in REPL? The context menu should also be updated to reflect

carlosperate, 2016-10-11:
Is there any chance to add  a `QKeySequence` to the context menu `addAction()` calls in #167 ? Ideally it would show up like most menu item shortcuts do, and that could also indicate Windows users of the different key shortcut.

[image]

ntoll, 2018-06-04:
This has been fixed by SHIFT-CTRL-[C|V] etc...

closed 2018-06-04 by ntoll

carlosperate, 2018-06-04:
Could we get the SHIFT-CTRL-[C|V]  in the documentation if it's not there already?

ntoll, 2018-06-04:
It'll be in the keyboard shortcuts tutorial on the new website. ;-)

carlosperate, 2018-06-04:
Awesome, thanks!

carlosperate, 2018-07-05:
> It'll be in the keyboard shortcuts tutorial on the new website. ;-)

I've created https://github.com/mu-editor/mu-editor.github.io/issues/30 to track things to include in this shortcuts page.