mu-editor/mu #170: Context Menu in REPL

eduvik, 2016-10-09:
The context menu in the REPL shows Undo, Redo and Cut and Delete. These are all meaningless in this context (pardon the pun) so should probably be removed. Would probably need to create a custom context menu instead, or the context menu could be disabled entirely.

ZanderBrown, 2016-10-10:
6c73e0150d73e86c87acdd951afda69f274dd494 in #167 

But you knew that already

askvictor, 2016-10-10:
The issue came before the commit :)

Copy-paste context menu would be nice, but disabling it is a short term fix

carlosperate, 2016-10-10:
I'm not quite sure how I feel about this one. When running a workshop a lot of kids copy/pasted the examples and instructions into the REPL by using the context menu, not sure they would all have been aware of how to use the keyboard shortcuts. I think it might be preferable to have meaningless options and maintain copy/paste, than to remove the entire menu.

askvictor, 2016-10-10:
Yeah I hear that. Perhaps toolbar buttons? Also, the delete ( and I think cut) options do actually remove text from the repl, which pane, which is pretty bad.

askvictor, 2016-10-10:
Wow; qt is so much easier than I expect https://github.com/mu-editor/mu/pull/167/commits/5b8716f2c3e056f616a56e1eecd610b5104e9f16

askvictor, 2016-10-10:
Eek; pastes the letters into the UI, but doesn't send them to the device :(

eduvik, 2016-10-10:
OK; fixed to some extent in https://github.com/mu-editor/mu/pull/167/commits/ab89cbe295b8b45458d390690c6ba47633c57428 - paste now pastes straight to the serial port. However, the displayed text is sometimes not what was pasted - for example, pasting 'information' displays 'onformation' even though the device receives 'information'. Might be a serial timing issue.

eduvik, 2016-10-10:
Now this is weird; even adding a small delay between sending each character, the problem persists. The first character of each paste seems to get mangled; sometimes in different ways. Seems to be length dependant too; sending 'inf' seems to be OK, but 'info' comes back as 'onfo'. Not sure if this is at the device, the hardware, driver, or library level :-/

carlosperate, 2016-10-11:
> @carlosperate - how did kids paste into the REPL when it doesn't send the data to the device?

I wasn't seating with the kids when I saw them use the context menu, so I didn't really noticed if it worked or now, it's very likely that it didn't work and then they typed it.

About the weird first character issue, have you tried changing the encoding as well?

eduvik, 2016-10-11:
the character is incorrect before it's decoded from bytes. i.e. it comes into the 'serial' object wrong. The device receives it fine, so there's no problem encoding on the sending side.

eduvik, 2016-10-11:
Interestingly, there is no weird first character issue when using the ESP8266 on my multi-device fork. Since the receiving code is the same for both devices, I'm left to suspect the device or the driver.

ntoll, 2018-06-04:
Fixed. Closing. :-)

closed 2018-06-04 by ntoll