mu-editor/mu #171: Can't open help menu

titimoby, 2016-10-17:
From Mu 0.9.11, I click on Help icon
Nothing happens but the terminal shows the following error:

`/usr/lib/firefox/firefox: /tmp/_MEIvIjOGD/libstdc++.so.6: version 'GLIBCXX_3.4.21' not found (required by /usr/lib/firefox/firefox)
`

ntoll, 2016-10-17:
Hmm.... your default browser appears to be Firefox right..? I suspect that it's not installed correctly since the error you report is a message from your OS that shows up in Mu's logs complaining about Firefox.

Essentially, Mu has asked the OS to start its default browser and point it at our online documentation. The OS is reporting that it can't start Firefox. Not a lot Mu can do about that... so can you check your browser configuration..?

titimoby, 2016-10-17:
My default browser is Firefox.
I can check ... but don't know what to search for.
All others applications i have that launch browser for help pages seems to work.
If you have any clue, I'll check. I need to understand it because I have a workshop with kids next sunday and I only have access to the machines at  that moment.
Thanks

carlosperate, 2016-10-17:
See if something like this could fix it: http://askubuntu.com/questions/575505/glibcxx-3-4-20-not-found-how-to-fix-this-error
Can you run firefox from the command line?

ntoll, 2016-10-23:
@titimoby any progress on this?

titimoby, 2016-10-23:
No progress on this topic, which I understand is not a problem from Mu editor.
I just come back from a workshop were I used Mu with kids and it was great.
I need now to find enough time to contribute to a french documentation to reach younger ones.

carlosperate, 2017-09-27:
@titimoby sounds like the original issue was with firefox instead of Mu, so I'll close this one. Feel free to open it again otherwise.

closed 2017-09-27 by carlosperate