mu-editor/mu #173: Tries to flash onto wrong drive

freddinator, 2016-10-20:
Mu tries to flash onto the D: drive, but my microbit is mounted on the H: drive. It eventually work, but pops up with "There is no disk in the drive. Please insert a disk into drive D:.". It's actually impossible for the microbit to be on Drive D:, as this is a reserved name used for system recovery.

ntoll, 2016-10-20:
Hi @freddinator - sorry to hear you're having problems. 

Can you please provide details of the OS you're using (by the looks of it some version of Windows).

To be clear, Mu works on top of the uFlash module which detects the micro:bit by looking for a drive with the name "MICROBIT". :-)

Do you have any non-standard settings in your Windows setup? Mu (and uFlash) are only following the directions given to them by the host operating system. 

If you're interested, the specific bit of code that does this is found here: https://github.com/ntoll/uflash/blob/master/uflash.py#L174 As you'll see uFlash tries all the drive letters (A->Z) and returns details about the first one it finds that contains the name MICROBIT.

Does this make sense? Basically, I'll need to know if you've something non-standard about your Windows OS. This is the first time anyone has had this problem... so it's, er, intriguing. ;-)

freddinator, 2016-10-20:
Yes, it's Windows 10. I ran the relevant sections of the code separately through command line python and it correctly produces "H:\". It does actually flash correctly, so I think that for some reason checking if the path exists or getting the volume name is attempting to read it, producing the error.. I do have a hard drive partition for my linux dualboot, but that isn't mounted as a drive letter.

ntoll, 2016-10-20:
Hmmm.... OK. I wonder how best to recreate it so it can be fixed. I don't own a Windows 10 machine so I can't test things. If you're feeling brave you're more than welcome to try yourself and submit a pull request to uFlash (N.B. I'm not trying to dodge a bullet here ~ rather, invite you to have a play around and, if possible, submit more information and a patch). Fame, fortune and, er, your name in the AUTHORS file awaits... :-)

Alternatively, I'll ask @tjguk (who is a bit of a Windows whiz) to see if he can reproduce / see what the problem is.

In any case, many thanks for contributing to the project just by bringing our attention to this problem! :-)

freddinator, 2016-10-20:
I've been able to confirm that running `os.path.exists("D:\\")` returns false, but produces the "There is no disk in drive" message. It also does this for drives E and F.

ntoll, 2016-10-20:
Hmmm... this is starting to sound familiar - I think we may have seen this once before. I'll check the bug history for Mu and uFlash to see and reference the issue here.

ntoll, 2016-10-20:
I was right... see https://github.com/ntoll/uflash/issues/10 in the uflash repos. Looks like we didn't work out what to do back then. @tjguk any suggestions?

If not, I'll do some research over the weekend and see if I can make it work in a virtualised version of the OS (not ideal though).

freddinator, 2016-10-20:
I've looked into this issue, and have found something which may be useful.
```python
os.system("vol %s: 2>nul>nul" % ("D")) == 0 and get_volume_name("D:\\") == 'MICROBIT'
```
That returns false with no error, and replacing it with drive H (the microbit) returns true.

ntoll, 2016-10-20:
@freddinator you star! I'll have a play at the weekend and (hopefully) get the fix in. :-)

Do you want to be added to the AUTHORS file for uflash..?

freddinator, 2016-10-20:
Sure! Glad I could help!

tjguk, 2016-10-21:
(Adding directly on github as emails appear not to have got through)

Sorry; rather late to the party. In principle we ought to be able to call SetErrorMode with SEM_FAILCRITICALERRORS before getting the volume information and then resetting afterwards.
https://msdn.microsoft.com/en-us/library/ms680621(v=vs.85).aspx
I've tried two or three machines with different SD-slot setups, and none of them fail in the way described, so I'm not very able to test. I'll try to run up a test script based on the current uflash code to see if the OP can confirm.

@ntoll: should this be targetting the uflash repo itself, or the version in mu/contrib?

tjguk, 2016-10-21:
(Adding directly on github as emails appear not to have got through)

I've attached a simple version of the find-microbit code, snipped from the current uflash codebase and with the SetErrorMode logic added. Where I am now, I have neither a micro:bit, nor a failing drive letter, but I'm fairly confident it should do what's needed. If someone could test, please, it should find the micro:bit or raise an exception, without throwing up the "No disk in the drive" dialog box. 
[get_volume_name.txt](https://github.com/mu-editor/mu/files/544474/get_volume_name.txt)

tjguk, 2016-10-22:
I've issued a PR against the upstream uflash repo referencing this issue and issue10 on that repo.

ntoll, 2016-10-23:
This should be fixed in the latest release.

closed 2016-10-23 by ntoll

tjguk, 2016-10-25:
(I'm not sure what github does with an attachment: hopefully it'll turn 
into a gist or something).

I've attached a simple version of the find-microbit code, snipped from 
the current uflash codebase and with the SetErrorMode logic added. Where 
I am now, I have neither a micro:bit, nor a failing drive letter, but 
I'm fairly confident it should do what's needed. If someone could test, 
please, it should find the micro:bit or raise an exception, without 
throwing up the "No disk in the drive" dialog box.

import os
import ctypes

def get_volume_name(disk_name):
    """
    Each disk or external device connected to windows has an attribute
    called "volume name". This function returns the volume name for
    the given disk/device.

    Code from http://stackoverflow.com/a/12056414
    """
    vol_name_buf = ctypes.create_unicode_buffer(1024)
    ctypes.windll.kernel32.GetVolumeInformationW(
        ctypes.c_wchar_p(disk_name), vol_name_buf,
        ctypes.sizeof(vol_name_buf), None, None, None, None, 0)
    return vol_name_buf.value

SEM_FAILCRITICALERRORS = 0x0001
old_mode = ctypes.windll.kernel32.SetErrorMode(SEM_FAILCRITICALERRORS)
try:
    for disk in 'ABCDEFGHIJKLMNOPQRSTUVWXYZ':
        path = '{}:\\'.format(disk)
        if os.path.exists(path) and get_volume_name(path) == 'MICROBIT':
            print("Microbit found at", path)
            break
    else:
        raise RuntimeError("Microbit not found")
finally:
    ctypes.windll.kernel32.SetErrorMode(old_mode)