mu-editor/mu #174: Add 'open file in editor' option to file on host drive FILES view

whaleygeek, 2016-10-22:
It would be helpful to be able to click on the file in the right hand pane in the FILES view and open it to edit the host version of that file in the default editor for that file. 

e.g. if you have a text file with some data in it, instead of having to find out where Mu stores those files (userdir/python) and navigating there to edit it, you could just double click on it and open it in a text editor. I'd be happy if this action was limited to just .txt and .py files for example, and if it allowed that file to be opened in a new tabbed pane in the Mu editor.

carlosperate, 2016-10-27:
Yeah, I completely agree this would be the expected behaviour by most people.

I would probably go as far as to suggest to be able open the microbit files as well, but in the spirit of keeping things simple I'd probably suggest to instead open a message box asking if you'd like to copy the file into the computer first and open from there (so clicking "OK" would do the copy and opening).

whaleygeek, 2017-02-01:
@carlosperate @ntoll is this on the radar at all do you know?

ZanderBrown, 2017-02-01:
I'll have a look into this later as it shouldn't be too difficult to achive

DavidWhaleMEF, 2018-06-22:
Super, thanks @ZanderBrown this is ace!

PR #499 (2018-07-07): Context menu for local files in micro:bit mode
Finally resolve #174, when I said I'd look into it later I had intended a few days, not 16 months! (sorry @whaleygeek)

All local files gain an "Open" command which basically gives the file to Qt to *do the right thing™*
[image]

Files with a `.py` or `.hex` extension have an additional "Open in Mu" which *does what it says on the tin™*
[image]

And I was right, it wasn't too difficult! (apart from a moment of silliness with the tests)
closed 2018-07-07 by ntoll