mu-editor/mu #175: sing bug fix (invalid phoneme) feature is not integrated into latest Mu

whaleygeek, 2016-10-22:
Could you integrate the bug fix for sing phoneme error handling into the latest Mu?

https://github.com/bbcmicrobit/micropython/issues/351

ZanderBrown, 2016-10-22:
So what's needed is for uFlash to be updated upstream against the latest MicroPython then pulled into Mu.

ntoll, 2016-10-23:
@whaleygeek I hear you.

closed 2016-10-23 by ntoll

whaleygeek, 2016-10-23:
@ntoll I'm sure you do :) I was just trying to follow the process, so it doesn't get missed off. 

I've reported a few things via email that have been missed off, so I thought I would be a bit more 'open source' and log it as an issue so it can be assigned to the next release. 

I was a little surprised you closed this issue, unless it is now fixed and deployed on the live website?
Is it possible you could re-open this issue until the new version is deployed with this fix in it, and then close it (as I am monitoring this issue as a trigger to report to the original requester that the fix is now deployed, and I nearly got a bit trigger happy when I saw this closed to tell them it was fixed!!)

Thanks :)

David

ntoll, 2016-10-23:
I'm about 80% there to a release. I closed it because I was already aware (!!) and I'm currently updating some stuff so the release is on the main website.

ntoll, 2016-10-23:
I.e. the thing is all built and so on... I'm crossing the Is and dotting the Ts. :-P