mu-editor/mu #177: Signed copy for Mac App store

whaleygeek, 2016-10-27:
microbit.org-ticket: 154

A customer has requested whether it would be possible to provide a signed version of the Mac version of Mu via the app store - the only way to install Mu on a Mac is to turn off the security settings when installing it.

I understand from talking to @ntoll that the dev team are aware of this, but it's more to do with priorities and finding people with some time to manage the Apple process.

But I wanted to raise this issue so that it is visibly on the todo list.

carlosperate, 2016-10-27:
There is already issue https://github.com/mu-editor/mu/issues/148 covering signing on OS X.

whaleygeek, 2017-02-01:
The related ticket #206 was closed down, so I see no mechanism that this ticket will be progressed any further.

closed 2017-02-01 by whaleygeek

carlosperate, 2017-02-01:
#206 is closed, but #148 is still open to track this.