mu-editor/mu #178: Better compatibility for restricted environments in schools

rpiuser, 2016-10-29:
Would there be any possibility of setting a default save / load path under a windows environment ? Currently C: is restricted and My Documents is redirected using GPO. When loading a "This operation has been cancelled due to restrictions in effect on this computer" message is displayed (Due to C: being restricted) clicking okay takes you to "My Documents" The same behaviour happens with save. Ideally pointing to the users home folder (e.g H:\Mu) would be a lot cleaner and a much better experience / less confusing for the students using it.

ZanderBrown, 2016-10-29:
Take a look at #126 & #86

rpiuser, 2016-10-29:
Thank you, I've read through the threads and can't seem to determine if this has been resolved. I'm assuming I need the settings.json and I need to set a path in there, however I'm unsure of the format of the path  :- "paths": ["path/foo.py" so would I use "paths": ["path/H:/mu"] ? I'll try testing.

ZanderBrown, 2016-10-29:
Ah yes seems that didn't happen. I'm assuming your using the exe?

rpiuser, 2016-10-29:
Using this build mu-2016-10-23_16_42_53.exe on windows

ZanderBrown, 2016-10-29:
Okay. Having received some support for the feature in #179 hopefully it should make it into master over the next week to be included in the next release

rpiuser, 2016-10-30:
Thank you, much appreciated

ZanderBrown, 2016-11-08:
#179 has been merged. Does it resolve your issue?

carlosperate, 2016-11-09:
As a general note to remind us before we close this issue: We really need to add a page about the settings file to the documentation so that this feature can be discovered.

rpiuser, 2016-11-10:
Using mu-0.9.13.win works great thank you, however if you specify a wrong drive letter or the drive doesn't exist in settings.json mu exits with a run returned -1

ZanderBrown, 2016-11-10:
ah makes sense. I will put a check in asap

rpiuser, 2016-11-10:
Other than that, it's working great - kids are merrily coding away using it :)

ntoll, 2018-06-04:
Closing since fixed.

closed 2018-06-04 by ntoll