mu-editor/mu #181: On raspbian, clicking the Files panel adds a new panel every time

carlosperate, 2016-10-30:
Saw this behaviour a few weeks ago at a Raspberry Jam, but since I was running a test version of Mu I wanted to replicate the issue on master and get specific versions before posting it here.

Basically, on raspbian clicking on the Files button will add a panel on every click:
[image]

Tested on 3.4.2
```
pep8==1.5.7
pyflakes==0.8.1
pyserial==2.6
```

```
python3-pyqt5:
  Installed: 5.3.2+dfsg-3+b1
  Candidate: 5.3.2+dfsg-3+b1
  Version table:
 *** 5.3.2+dfsg-3+b1 0
        500 http://mirrordirector.raspbian.org/raspbian/ jessie/main armhf Packages
        100 /var/lib/dpkg/status

python3-pyqt5.qsci:
  Installed: 2.8.4+dfsg-1
  Candidate: 2.8.4+dfsg-1
  Version table:
 *** 2.8.4+dfsg-1 0
        500 http://mirrordirector.raspbian.org/raspbian/ jessie/main armhf Packages
        100 /var/lib/dpkg/status

python3-pyqt5.qtserialport:
  Installed: 5.3.2+dfsg-3+b1
  Candidate: 5.3.2+dfsg-3+b1
  Version table:
 *** 5.3.2+dfsg-3+b1 0
        500 http://mirrordirector.raspbian.org/raspbian/ jessie/main armhf Packages
        100 /var/lib/dpkg/status
```

Could not replicate on x86 Ubuntu 14.04  Python 3.4.3:

```
pep8==1.7.0
pycodestyle==2.0.0
pyflakes==1.2.3
pyserial==3.1.1
```

```
python3-pyqt5:
  Installed: 5.2.1+dfsg-1ubuntu1
  Candidate: 5.2.1+dfsg-1ubuntu1
  Version table:
 *** 5.2.1+dfsg-1ubuntu1 0
        500 http://gb.archive.ubuntu.com/ubuntu/ trusty/main amd64 Packages
        100 /var/lib/dpkg/status

python3-pyqt5.qsci:
  Installed: 2.8.1-2ubuntu1
  Candidate: 2.8.1-2ubuntu1
  Version table:
 *** 2.8.1-2ubuntu1 0
        500 http://gb.archive.ubuntu.com/ubuntu/ trusty-updates/universe amd64 Packages
        100 /var/lib/dpkg/status
     2.8.1-2 0
        500 http://gb.archive.ubuntu.com/ubuntu/ trusty/universe amd64 Packages

python3-pyqt5.qtserialport:
  Installed: 5.2.1+dfsg-1ubuntu1
  Candidate: 5.2.1+dfsg-1ubuntu1
  Version table:
 *** 5.2.1+dfsg-1ubuntu1 0
        500 http://gb.archive.ubuntu.com/ubuntu/ trusty/main amd64 Packages
        100 /var/lib/dpkg/status
```

ZanderBrown, 2016-10-30:
This also occurs in my branch (#179)

EVERY click on the files button causes the following error (even on the initial click)
```
Traceback (most recent call last):
  File "/usr/local/lib/python3.4/dist-packages/mu-0.9.12-py3.4.egg/mu/logic.py", line 464, in toggle_fs
    self.add_fs()
  File "/usr/local/lib/python3.4/dist-packages/mu-0.9.12-py3.4.egg/mu/logic.py", line 434, in add_fs
    self._view.add_filesystem(home=get_python_dir())
  File "/usr/local/lib/python3.4/dist-packages/mu-0.9.12-py3.4.egg/mu/interface.py", line 663, in add_filesystem
    self.fs = FileSystemPane(self.splitter, home)
  File "/usr/local/lib/python3.4/dist-packages/mu-0.9.12-py3.4.egg/mu/interface.py", line 1092, in __init__
    self.ls()
  File "/usr/local/lib/python3.4/dist-packages/mu-0.9.12-py3.4.egg/mu/interface.py", line 1102, in ls
    microbit_files = microfs.ls(microfs.get_serial())
  File "/usr/local/lib/python3.4/dist-packages/mu-0.9.12-py3.4.egg/mu/contrib/microfs.py", line 135, in ls
    ], serial)
  File "/usr/local/lib/python3.4/dist-packages/mu-0.9.12-py3.4.egg/mu/contrib/microfs.py", line 94, in execute
    raw_on(serial)
  File "/usr/local/lib/python3.4/dist-packages/mu-0.9.12-py3.4.egg/mu/contrib/microfs.py", line 60, in raw_on
    serial.read_until(b'\n>')  # Flush buffer until prompt.
AttributeError: 'Serial' object has no attribute 'read_until'
```

ZanderBrown, 2016-10-30:
Running `sudo pip3 install pyserial --upgrade` thus upgrading PySerial from `2.6` to `3.2` fixes the issue however `2.6` is the latest available through apt.

ntoll, 2016-10-31:
Hmm... nice find. 

I believe we should do a better job of our requirements.txt file and pin versions. That would have mitigated the above given the Raspbian packager would have been warned of the version mismatch (I presume).

I'll ping someone at the RPi Foundation about this...

ZanderBrown, 2016-10-31:
apt would complain about the missmach but because of the Debian update policy there isn't a newer version that apt knows about so it would refuse to install mu

Also `requirements.txt` is only useful when installing through pip which does know about the newer PySerial.

Hope that makes some sense.

ntoll, 2016-10-31:
Yes it does. What I meant was that I believe the Debian packaging looks at our requirements.txt and *at that point* would complain before building the package.

ZanderBrown, 2016-10-31:
it *might* look at `requirements.txt` but that wouldn't help that `'Serial' object has no attribute 'read_until'` in the latest PySerial avalible on Debian/Raspbian through apt.

Maybe we could use an alternate function?

ntoll, 2018-06-04:
This has been fixed. Closing. :-)

closed 2018-06-04 by ntoll