mu-editor/mu #182: Mac OSX: qscintilla2.rb fails to install with xcode 8

tdicola, 2016-10-31:
There is an issue with the latest version of xcode (8.0) and the qscintilla2.rb homebrew file.  See this issue for more context and discussion: https://github.com/paralect/robomongo/issues/1220 and https://github.com/Homebrew/homebrew-core/pull/5877

As a workaround brew install the updated qscintilla2.rb from 5877 above:
```
brew install brew install https://raw.githubusercontent.com/ilovezfs/homebrew-core/66311089890da0ee13ae47f39542c7451754bd1e/Formula/qscintilla2.rb
```

ntoll, 2016-10-31:
Hey @inglesp do you have any opinion / visibility on this..?

carlosperate, 2016-10-31:
We've been using this formula [package/extras/qscintilla2.rb](https://github.com/mu-editor/mu/blob/master/package/extras/qscintilla2.rb) (instructions in the [package/README.md](https://github.com/mu-editor/mu/blob/2fa0ab4a3cdf311748dad34e75e4cce0f1a18d56/package/README.md)) for the Travis CI installation, not quite sure if for a similar reason.
Maybe we should revisit as well to see if it needs updating.

carlosperate, 2018-03-17:
Thanks for the original report, since we are now installing all dependencies via PyPI we don't have this issue any more.

closed 2018-03-17 by carlosperate