mu-editor/mu #183: Accessibility - add keyboard short-cuts for the buttons

ntoll, 2016-11-03:
I've been liaising with people who are interested in having a Braille display work with Mu. It would be *really* useful for them if we had keyboard short-cuts for the buttons. They suggest the following schema:

* New - `Ctrl+N`
* Load - `Ctrl+O`/`Ctrl+L`
* Save - `Ctrl+S`
* Flash - `F9`
* Files - `Ctrl+F`
* Repl - `Ctrl+R`
* Zoom-In - `Ctrl+Shift+plus`
* Zoom-Out - `Ctrl+Shift+minus`
* Theme - `Ctrl+T`
* Check - `F8`
* Help - `Ctrl+H`
* Quit - `Alt+F4`

ZanderBrown, 2016-11-03:
Looks good but:

Files `Ctrl+F` - This is an editor and most editors use `Ctrl+F` for 'Find' which maybe confusing

Help `Ctrl+H` - Most programs use `F1` for help and editors often use this for 'Replace'

carlosperate, 2016-11-03:
This should be a useful guideline: https://en.wikipedia.org/wiki/Table_of_keyboard_shortcuts

scruss, 2017-06-23:
Doesn't use the standard system quit of Ctrl+Q or Ctrl+W

ntoll, 2018-06-04:
Implemented (ages ago) as per the helpful guidelines and pointers from everyone. Thank you!

closed 2018-06-04 by ntoll