mu-editor/mu #184: REPL and Files sometimes don't work on Windows 10 Preview (USB-serial)

tsh-xx, 2016-11-08:
I'm running windows 10 preview, and have problems connecting the the USB-Serial device on the microbit. I'm not sure if this is a specific problem with my machine, or specific to the preview builds. It also seems to be lower-level than just Mu (putty doesn't work, teraterm only works occasionally) - but it may help someone to at least log it here.
The problem exists with both the mbed serial driver, and the native Windows10 usb-serial driver, but only using the USB ports (both USB2 and USB3) provided by my motherboard chipset. 

Using a ' Inateck PCI-E to USB 3.0 2-Port' card, the interface seems to work as expected (Files and REPL function normally).

With the non-working interface, the serial port appears in device manager. When attempting to open files or REPL, mu hangs for many seconds before popping up a warning that it was unable to configure the port (similar to the failure mode in Putty).

carlosperate, 2016-11-08:
Hi @tsh-xx,

As I mentioned on the Slack channel I've created this console version of Mu, so hopefully with this version you could get the log location and we could have a better look at why it might be failing.

https://s3-us-west-2.amazonaws.com/ardublockly-builds/microbit/windows/mu-2016-11-08_12_23_50.exe

carlosperate, 2016-11-16:
hi @tsh-xx, did you have a chance to have a play with the console version of Mu?

tsh-xx, 2016-11-16:
@carlosperate Just tried it, and I get no console output at all from that exe.

carlosperate, 2016-11-16:
When executing this .exe (just by double clicking) there should be two windows opening, one a normal windows console, and the other the normal Mu GUI, does that not happen?

tsh-xx, 2016-11-17:
OK, that works (just not running it from a command line). The log file is generated, but doesn't seem to show anything very interesting. What is more interesting, now if i wait for it to time-out, the 2nd time round, the connection succeeds, and I can use REPL and files normally. Un-pluging the micro-bit resets this state. I'm not sure if I didn't notice this last week or if there has been an OS update of the drivers.

The controller which shows this behaviour is _Eltron USB 3.0_, the other controller which seems to work OK is _Fresco Logic USB 3.0_

[mu.log.txt](https://github.com/mu-editor/mu/files/596840/mu.log.txt)

ntoll, 2018-06-04:
This appears to be fixed on Windows 10 (for me).

closed 2018-06-04 by ntoll