mu-editor/mu #186: OSX quirks

jarjartee, 2016-11-13:


jarjartee, 2016-11-13:
I’ve been trying flash a script from ‘mu’ on to microbit ; loads 98% correct (as in when connected to a ‘BOT’) but with a few errors. This does NOT happen when I use the ONLINE version i.e drag file over to microbit ; then 100% correct. Now using El Capitan OSX with latest mu release;

# Add your Python code here. E.g.
from microbit import *

leftA = pin1
leftB = pin12
rightA = pin0
rightB = pin8

def forward(speedpct):
    speed = speedpct * 10.23
    leftA.write_analog(speed)
    leftB.write_digital(0)
    rightA.write_analog(speed)
    rightB.write_digital(0)

def reverse(speedpct):
    speed = speedpct * 10.23
    leftA.write_analog(1023-speed)
    leftB.write_digital(1)
    rightA.write_analog(1023-speed)
    rightB.write_digital(1)

def spinRight(speedpct):
    speed = speedpct * 10.23
    leftA.write_analog(1023-speed)
    leftB.write_digital(1)
    rightA.write_analog(speed)
    rightB.write_digital(0)

def spinLeft(speedpct):
    speed = speedpct * 10.23
    leftA.write_analog(speed)
    leftB.write_digital(0)
    rightA.write_analog(1023-speed)
    rightB.write_digital(1)

def stop():
    leftA.write_digital(0)
    leftB.write_digital(0)
    rightA.write_digital(0)
    rightB.write_digital(0)
    
while True:
    display.show(Image.ARROW_N)
    forward(50)
    sleep(8000)
    display.show(Image.ARROW_E)
    spinRight(20)
    sleep(8000)
    display.show(Image.ARROW_S)
    reverse(50)
    sleep(5000)
    display.show(Image.ARROW_W)
    spinLeft(100)
    sleep(5000)
    display.show(Image.HAPPY)
    
    stop()
    sleep(3000)

jarjartee, 2016-11-13:
tried new and edited code ; new definition for 'stop' - now works on mu ;
# Add your Python code here. E.g.
from microbit import *

leftA = pin1
leftB = pin12
rightA = pin0
rightB = pin8

def forward(speedpct):
    speed = speedpct * 10.23
    leftA.write_analog(speed)
    leftB.write_digital(0)
    rightA.write_analog(speed)
    rightB.write_digital(0)

def reverse(speedpct):
    speed = speedpct * 10.23
    leftA.write_analog(1023-speed)
    leftB.write_digital(1)
    rightA.write_analog(1023-speed)
    rightB.write_digital(1)

def spinRight(speedpct):
    speed = speedpct * 10.23
    leftA.write_analog(1023-speed)
    leftB.write_digital(1)
    rightA.write_analog(speed)
    rightB.write_digital(0)

def spinLeft(speedpct):
    speed = speedpct * 10.23
    leftA.write_analog(speed)
    leftB.write_digital(0)
    rightA.write_analog(1023-speed)
    rightB.write_digital(1)

def stop():
    leftA.write_analog(0)
    leftB.write_digital(0)
    rightA.write_analog(0)
    rightB.write_digital(0)
    
while True:
    display.show(Image.ARROW_N)
    forward(50)
    sleep(8000)
    display.show(Image.ARROW_E)
    spinRight(20)
    sleep(8000)
    display.show(Image.ARROW_S)
    reverse(50)
    sleep(5000)
    display.show(Image.ARROW_W)
    spinLeft(100)
    sleep(5000)
    display.show(Image.HAPPY)
    
    stop()
    sleep(3000)

ntoll, 2018-05-12:
My guess is that it may be something to do with the versions of the MicroPython runtime bundled with Mu / browser based editor.

I'll close anyway. @jarjartee, please feel free to re-open if the problem still exists.

closed 2018-05-12 by ntoll