mu-editor/mu #187: Windows packaging instruction issues

tdicola, 2016-11-14:
I tried to follow the instructions to build a packaged executable of Mu on Windows from here https://github.com/mu-editor/mu/blob/master/package/README.rst but ran into a lot of trouble.  To help folks that hit similar issues, here are the main problems:

-   You probably want to clarify that most people should use 32-bit python.  If you use 64-bit python you'll build a version of mu that only works on 64-bit Windows whereas 32-bit apps will run on both 32-bit and 64-bit windows.  From your appveyor config for making official mu packages it appears you're using 32-bit python.

-   Don't install python 3.5 or higher, you want python 3.4.x.  The reason for this is that the PyQt5 binary installer from Riverbank software doesn't exist anymore for the latest PyQT versions.  Instead Riverbank seems to want people to install with pip, but pip installed PyQT binaries seem to be incompatible with packaging releases using pyinstaller (I tried and it just fails to find all the PyQT DLL files to copy into the package).  The reason you want Python 3.4.x is that you want to use PyQt 5.4.2 binaries which are only available for Python 3.4.x builds.  Find this installer here (again this is what the appveyor config seems to use): https://sourceforge.net/projects/pyqt/files/PyQt5/PyQt-5.4.2/

- Don't pip install pyinstaller, instead install exactly version 3.1.1 using `pip3 install pyinstaller==3.1.1`.  Again the appveyor config mentions and does this, the issues seems to be a new requirement on VC++ redistributables in later pyinstaller versions (I ran into this issue too, I think it needs VC++ 2010 redistributables if anyone has issues).

Once you have all those pieces lined up building a Windows package seems to work fine.  In general anyone attempting to do this should really just ignore the packaging instructions and look at the appveyor config in the project root--this appears to be the source of truth for how to package mu on windows.  IMHO it's worth mentioning in the packaging readme for the Windows instructions.

ntoll, 2016-11-14:
@tdicola thanks for this!n @carlosperate (our packaging supremo) any thoughts on the above..? The lack of / incompleteness of PyQT binaries for Python 3.5 is concerning.

I take your point about the README.

carlosperate, 2016-11-14:
Hi @tdicola,

Thanks for pointing this out, you are correct in all your points, we've left the package README file a little bit behind. I'll look at #188 , but yeah we should include all this in the instructions.

I'll have a look at using Pyinstaller with PyQt5.7 from the pip wheels, and check if this is something we could work around with the spec file and/or pyinstaller hooks. From the sounds of it the obvious workaround might be to identify and add the dll paths, but it could end up being a bit of a fragile workaround.

carlosperate, 2016-11-18:
Another update, as the situation is not looking all that good. On windows there seems to be a problem with Python 3.5 and Pyinstaller build on an environment with the Universal CRT? (it's not yet clear to me the exact trigger for this issue but whatever it is, it happens in our AppVeyor environment).

After a small update with the pyinstaller spec file it will package and run with Py3.5 and PyQt/Qscintilla using the PyPI wheels, but it might not run correctly on Windows versions < 10 that do not have a specific windows update ([KB2999226](https://support.microsoft.com/en-gb/kb/2999226)), visual studio 2015 installer, the 2015 VC++ redistributable, or any of the other ways to get the Universal CRT installed.

https://github.com/pyinstaller/pyinstaller/issues/1566

This is not an issue with Python 3.4, so we can still safely use the old build method.

I am not yet sure if there is a way to include enough dlls with the PyInstaller packaged application to fix this, so I will have to investigate further.

ntoll, 2018-06-04:
Closing since the above is obsoleted by new Windows packaging work.

closed 2018-06-04 by ntoll