mu-editor/mu #192: Idea for saving some script space

whaleygeek, 2016-11-23:
Nicholas and I met 'in a pub' for 'a pythonic meet up' last night. I forget who had the idea (we might have had it simultaneously, there was definitely a synchronised 'light bulb moment'!!

Regarding squeezing more python script source code into the limited 8K space on the micro:bit - the idea was that if you look at a typical script, where does a sizeable portion of the flash go to, and is there any low-hanging-fruit to save a bit of space by simple script compression?

Then we realised that there are a lot of spaces in Python source - each indent will probably be 4 spaces, and more than half of a typical large script will have at least one indent, possibly two or three, at the start of most lines. Each indent takes 4 spaces (so 4 bytes of flash).

e.g. in this 217 line script: https://github.com/whaleygeek/microbit_python/blob/master/connect4x4/connect4x4.py

There are easily 143 lines with an indent, and many lines with 2 or 3 levels of indent.

So, the idea would be for the editor (e.g. Mu) to compress all indent spaces at the left of any line to 1 space per indent - it would still be syntactically correct and semantically identical to the Python interpreter running on the micro:bit.

Even at 4 chars per indented line, that would be a saving of 429 bytes. But it is better than that, as many lines have multiple indents (and each additional compressed indent level will save another 3 bytes of flash).

If a .hex is opened by Mu or other editor, it could turn indent spaces at the start of the line back to multiple spaces when re-loading into the editor.

As a nice touch, there could be an option to put a magic comment at the top of the file to specify the preferred indent size, so that when the editor re-loads any script, it would know the user preference for spaces per indent.

Clearly it would still have to work with loading a .hex file that has a Python script with multiple spaces per indent.

But in the common case, quite a few bytes of flash could be stored in the persisted version of the script, for what looks like a relatively simple algorithmic compression process that would easily be possible inside the editor (and also inside a web editor) when serialising the script into a .hex. Before hexlifying the script, it could be put through an indent compression step. The compression algorithm is probably streamable, meaning you could probably compress and hexlify 'on the fly'.

carlosperate, 2016-11-23:
Sounds like an interesting idea to fit larger scripts. Couple of thoughts:
* We need to be careful about calculating the number of spaces per indentation, quite often I’ve seen learners using odd numbers of spaces, so we need to take that into account or we’ll introduce errors. 
* As sacrilegious as it might sound, I would prefer to replace spaces with tabs. In the case the script is extracted using a tool without this translation, then a tab is probably better than a single space.
* Since we are already checking for the file to be larger than the available memory, I’d suggest to only implement such functionality if this limit is being reached, as there is no need otherwise.

ZanderBrown, 2016-11-24:
+1 for use of tab

whaleygeek, 2017-02-01:
@ZanderBrown @carlosperate  Any progress on this issue at all? A number of users tell me that they have to take comments out of their Python programs in order to get them to fit. We work very very hard in schools to try to get kids to add comments to their program - then telling them to take them out again is just undoing all of that hard work! A bit more breathing space would help, and this would save 3 bytes of flash in the 8K region for every indent on every line, giving more space for comments and more code.

ZanderBrown, 2018-03-04:
One of the concerns with this was a miss match between error line number and source as a result of stripping comments.

It just occurred to me we could replace block comments with `\n` thus reducing the file size whilst maintaining line counts

So in summary what I believe we want is:

* Complete removal of inline comments
* Block comments (both `#` and `"""`) replaced with an appropriate number of `\n`
* Conversion of spaces to tabs
* striping extra whitespace from the end of each line

Whilst it won't work miracles I imagine this would free up a significant number of bytes especially in well-commented scripts

carlosperate, 2018-03-04:
That's a very good point about the mismatching code lines, good catch @ZanderBrown, +1 on replacing with new lines.

ZanderBrown, 2018-03-09:
Handling continuations is proving to be a pain

carlosperate, 2018-04-18:
Great news, @ZanderBrown and @ntoll have implemented a "minify" option for Python code flashed to a micro:bit (added in https://github.com/mu-editor/mu/commit/bf8bb7586e37f51e3f7434f17bb56e996eaf1f28), and this feature will be included in the next release :)

[image] 

I'll close this issue, but please feel free to reopen if you have any questions or suggestions.

closed 2018-04-18 by carlosperate