mu-editor/mu #193: Multiple FILE windows open when micro:bit not present

whaleygeek, 2016-11-23:
Mu version: ? (downloaded 24th Oct 2016) on Mac

I can get multiple file windows open sometimes when pressing the FILES button

[image]

This seems to be if you press FILES when the micro:bit is already running code so the REPL is not responding. I get a spinning beach ball mouse pointer when I press FILES, I unplug the micro:bit as that is the only way to unlock the Mac in this state. The beach ball stops spinning but the FILES window opens. I then plug in the micro:bit again, and press FILES, get spinning beach-ball mouse cursor, unplug USB again and I now have two FILES windows (and a very small editor window!) with no way to remove those FILES windows. I have to close Mu.

Note that I have a small looping program running on the micro:bit that auto starts when it boots.

carlosperate, 2016-11-23:
Seems somewhat related to https://github.com/mu-editor/mu/issues/193, but most likely to be a different issue. I had a quick look at the build environment for OS X and it packs pyserial 3.2.1.
Just to confirm, you are using the executable rather than the python sources, correct?

whaleygeek, 2016-11-23:
Thanks. Yes, I downloaded it on 24th Oct from codewith.mu and just used the binary.

ZanderBrown, 2016-11-24:
Pretty sure there is some relation to #181. Any logs?

whaleygeek, 2017-02-01:
Please can you send links to documentation that explain how to take a log from Mu.

whaleygeek, 2017-02-01:
I would like to be able to escalate valuable ticket reports into this repo, as per a recent conversation with @ntoll so I need the tools/information at hand so that we can do this.

ntoll, 2017-02-01:
When you start Mu from the command line, on all OS's it prints out were it's logging.

carlosperate, 2017-02-01:
> When you start Mu from the command line, on all OS's it prints out were it's logging.

That's only the case when running the Python code, or the executable in Linux. On Windows the .exe file doesn't have any console output, and on OSX you have to manually run the executable bundled inside the mu.app directory in `mu.app\Contents\MacOS\mu`.

ntoll, 2018-06-04:
Closing since fixed.

closed 2018-06-04 by ntoll