mu-editor/mu #195: Shortcut key to force 'locate microbit' dialogue

blogmywiki, 2016-12-03:
Would it be possible to add a keyboard shortcut to force Mu to locate / relocate the USB drive mapped to the Microbit?

This would be useful to me because I have problems with our school's VDI - somehow Mu (or the VDI) seems to 'forget' where the microbit is during a session even if it hasn't been unplugged. Mu says it is flashing the code but nothing happens, but closing Mu and relocating the device seems to fix it.

(It may also help when I trying to flash code to multiple microbits plugged in to different USB sockets.)

ZanderBrown, 2016-12-04:
This would appear to be something worth looking into

carlosperate, 2016-12-07:
Every time the "flash" button is pressed uFlash searches for a connected micro:bit, there is no caching for this information, so technically speaking there shouldn't be a difference by restarting the application.

If the window with the message that microbit is being flashed appears, then it must believe it has found the device correctly. Does the microbit LED not flash at all at this point? Is the micro:bit USB drive accessible?
An odd request, but when this happens, could you open the micro:bit drive on explorer and then try again from Mu?

It would also be quite useful if you could replicate this issue and then send us the `mu.log` log file.
On Windows you should be able to find it in `C:\Users\<YOUR_USERNAME>\AppData\Local\python\mu\Logs`, although I am not 100% if it will be the case as well on VDI connections.

ntoll, 2016-12-07:
@carlosperate I don't think that's the case. Take a look at https://github.com/mu-editor/mu/blob/master/mu/logic.py#L396 - I believe this is where the problem lies. When the device is flashed and restarts, I *suspect* that it's getting re-mounted elsewhere on the Windows file system (as another drive). However, given this is some sort of virtualised environment I can't tell for sure and don't want to speculate. In any case, when the student gets into this position, allowing them to re-set the user defined location of the device would probably be useful. It'd be good to look at the logs though, to see if this shines more light on the problem.

carlosperate, 2016-12-07:
You are right @ntoll, I think I misunderstood the question as I was only taking in consideration the auto-detection mechanism.
Yes, if the user selects a location manually that does get saved in a local variable that's reused in the future. My concern there is that this saved location also gets checked before flashing (although we should probably change it from `os.path.exists()` to `os.path.isdir()`), so unless the mapped drive is immediately mapped to something else I am unsure how this is happening (mapped drives that are no longer connected, but still appear in "my pc", do return False or at least my CD drive does).
Indeed I am quite curious to see this log file.

blogmywiki, 2016-12-15:
Apologies for delay - I won't have access to the equipment for a few weeks now, although I could try and replicate the issue using the same virtual machine but it would have to be using a MacBook as the client rather than the HP thin-clients we have in our IT suite.

ntoll, 2018-06-04:
Closing since this issue appears to no longer be a problem (feel free to re-open).

closed 2018-06-04 by ntoll