mu-editor/mu #197: setup.py fails on windows

ZanderBrown, 2016-12-07:
Ok so a bit of an unlikely use case i know but attempting to run `python setup.py install` on windows fails like so:

```
running install
running bdist_egg
running egg_info
creating mu.egg-info
writing mu.egg-info\PKG-INFO
writing entry points to mu.egg-info\entry_points.txt
writing dependency_links to mu.egg-info\dependency_links.txt
writing top-level names to mu.egg-info\top_level.txt
writing manifest file 'mu.egg-info\SOURCES.txt'
reading manifest file 'mu.egg-info\SOURCES.txt'
reading manifest template 'MANIFEST.in'
writing manifest file 'mu.egg-info\SOURCES.txt'
installing library code to build\bdist.win32\egg
running install_lib
running build_py
creating build
creating build\lib
creating build\lib\mu
copying mu\__init__.py -> build\lib\mu
copying mu\app.py -> build\lib\mu
copying mu\interface.py -> build\lib\mu
copying mu\logic.py -> build\lib\mu
creating build\lib\mu\contrib
copying mu\contrib\__init__.py -> build\lib\mu\contrib
copying mu\contrib\appdirs.py -> build\lib\mu\contrib
copying mu\contrib\microfs.py -> build\lib\mu\contrib
copying mu\contrib\uflash.py -> build\lib\mu\contrib
creating build\lib\mu\resources
copying mu\resources\__init__.py -> build\lib\mu\resources
copying mu\resources\api.py -> build\lib\mu\resources
creating build\lib\mu\resources\css
copying mu\resources\css\day.css -> build\lib\mu\resources\css
copying mu\resources\css\night.css -> build\lib\mu\resources\css
creating build\lib\mu\resources\fonts
copying mu\resources\fonts\LICENSE.txt -> build\lib\mu\resources\fonts
copying mu\resources\fonts\SourceCodePro-Bold.otf -> build\lib\mu\resources\fonts
copying mu\resources\fonts\SourceCodePro-BoldIt.otf -> build\lib\mu\resources\fonts
copying mu\resources\fonts\SourceCodePro-It.otf -> build\lib\mu\resources\fonts
copying mu\resources\fonts\SourceCodePro-Regular.otf -> build\lib\mu\resources\fonts
copying mu\resources\fonts\SourceCodePro-Semibold.otf -> build\lib\mu\resources\fonts
copying mu\resources\fonts\SourceCodePro-SemiboldIt.otf -> build\lib\mu\resources\fonts
creating build\lib\mu\resources\images
copying mu\resources\images\check.png -> build\lib\mu\resources\images
copying mu\resources\images\files.png -> build\lib\mu\resources\images
copying mu\resources\images\flash.png -> build\lib\mu\resources\images
copying mu\resources\images\help.png -> build\lib\mu\resources\images
copying mu\resources\images\icon.png -> build\lib\mu\resources\images
copying mu\resources\images\load.png -> build\lib\mu\resources\images
copying mu\resources\images\new.png -> build\lib\mu\resources\images
copying mu\resources\images\quit.png -> build\lib\mu\resources\images
copying mu\resources\images\repl.png -> build\lib\mu\resources\images
copying mu\resources\images\save.png -> build\lib\mu\resources\images
copying mu\resources\images\snippets.png -> build\lib\mu\resources\images
copying mu\resources\images\theme.png -> build\lib\mu\resources\images
copying mu\resources\images\theme_day.png -> build\lib\mu\resources\images
copying mu\resources\images\zoom-in.png -> build\lib\mu\resources\images
copying mu\resources\images\zoom-out.png -> build\lib\mu\resources\images
creating build\bdist.win32
creating build\bdist.win32\egg
creating build\bdist.win32\egg\mu
copying build\lib\mu\__init__.py -> build\bdist.win32\egg\mu
copying build\lib\mu\app.py -> build\bdist.win32\egg\mu
copying build\lib\mu\interface.py -> build\bdist.win32\egg\mu
copying build\lib\mu\logic.py -> build\bdist.win32\egg\mu
creating build\bdist.win32\egg\mu\contrib
copying build\lib\mu\contrib\__init__.py -> build\bdist.win32\egg\mu\contrib
copying build\lib\mu\contrib\appdirs.py -> build\bdist.win32\egg\mu\contrib
[... 54 lines cut ...]
    self.do_egg_install()
  File "C:\Users\<USER>\AppData\Local\Programs\Python\Python35-32\lib\site-packages\setuptools\command\install.py", line 109, in do_egg_install
    self.run_command('bdist_egg')
  File "C:\Users\<USER>\AppData\Local\Programs\Python\Python35-32\lib\distutils\cmd.py", line 313, in run_command
    self.distribution.run_command(command)
  File "C:\Users\<USER>\AppData\Local\Programs\Python\Python35-32\lib\distutils\dist.py", line 974, in run_command
    cmd_obj.run()
  File "C:\Users\<USER>\AppData\Local\Programs\Python\Python35-32\lib\site-packages\setuptools\command\bdist_egg.py", line 181, in run
    self.do_install_data()
  File "C:\Users\<USER>\AppData\Local\Programs\Python\Python35-32\lib\site-packages\setuptools\command\bdist_egg.py", line 133, in do_install_data
    self.call_command('install_data', force=0, root=None)
  File "C:\Users\<USER>\AppData\Local\Programs\Python\Python35-32\lib\site-packages\setuptools\command\bdist_egg.py", line 147, in call_command
    self.run_command(cmdname)
  File "C:\Users\<USER>\AppData\Local\Programs\Python\Python35-32\lib\distutils\cmd.py", line 313, in run_command
    self.distribution.run_command(command)
  File "C:\Users\<USER>\AppData\Local\Programs\Python\Python35-32\lib\distutils\dist.py", line 974, in run_command
    cmd_obj.run()
  File "C:\Users\<USER>\AppData\Local\Programs\Python\Python35-32\lib\distutils\command\install_data.py", line 56, in run
    dir = convert_path(f[0])
  File "C:\Users\<USER>\AppData\Local\Programs\Python\Python35-32\lib\distutils\util.py", line 125, in convert_path
    raise ValueError("path '%s' cannot be absolute" % pathname)
ValueError: path '/etc/udev/rules.d' cannot be absolute
``` 

It's seems quite a simple one really, we have hardcoded UNIX paths that of course don't work on window

KimHwaMok, 2017-03-07:
@ZanderBrown 
Hi, Did you solve this problem?
If you solved this problem, Can you tell me that how to install on Windows?

tjguk, 2017-03-07:
@KimHwaMok a simple ad hoc fix would be to remove the data_files parameter whose values here have no relevance on Windows. Longer term, it could be made conditional on the platform. Even longer term, the setup.py needs some love.

However... you don't actually need to install mu at all. You can just run it directly from its development directory (having first pip installed the requirements). Just: python run.py

tjguk, 2017-03-07:
I've created PR #257 as a quick fix for this if you're interested in testing

ntoll, 2017-03-07:
I don't have access to a Windows machine at the moment. If someone can confirm #257 works I'll merge it (and many thanks, as always, @tjguk).

ZanderBrown, 2018-04-07:
> If someone can confirm #257 works

*Opps*

However it seems the desktop file was removed from setup.py back here 1f0599ac69652640f96f625d2f24bf262c2489fc anyway making this and #257 irrelevant  although @tjguk's `gui_scripts` vs `console_scripts` may be worth investigating considering unlike the debug version `mu-editor` is supposed to run without a console

closed 2018-04-07 by ZanderBrown