mu-editor/mu #198: Including Custom Modules with Mu

armbennett, 2016-12-07:
Hello, to start with I just want to say how much I love Mu, it's working really well in class.

I'm working on a simple micro:bit controlled robot for my students to program and initially it would be great if they could import a custom module into Mu so they can use functions I have created to simplify the control of the robot before later getting them to 'look under the hood' and write their own functions.

Ideally I would like to write a module and build a custom Mu package that included it, however I'm not sure if this is possible or where to start. I know it is possible to manually copy additional files onto the micro:bit directly that can then be imported, but I would like to keep it as simple as possible for the students to begin with.

Any advice on how to get started would be much appreciated!

Thanks

Alex

ntoll, 2016-12-08:
Hi @armbennett,

To be honest, the simplest / easiest way to do this is to copy the module onto the device as you describe.

MicroPython itself has two ways to "bake in" modules into the runtime: frozen modules and native modules. I don't think frozen modules are enabled on the micro:bit version of MicroPython due to memory constraints. Put simply, frozen modules are your Python code but stored very efficiently as bytecode within the runtime - @tdicola has written a great overview here: https://learn.adafruit.com/micropython-basics-loading-modules/frozen-modules. The other way is to write a module in C and compile it in. This takes a lot of effort since you need to know C, know C well enough that you don't break things, and know C well enough to write it to work efficiently on such a memory constrained device. This is the approach we took for the micro:bit since the results are fast and take up less resources in terms of size.

Finally, assuming you manage to make your own build of MicroPython you'll need to create your own version of Mu to use your new runtime. A handy alternative is to use uflash - the module Mu uses to flash the device - it has "power user" options to allow you to specify bespoke versions of the runtime. Documentation here: https://uflash.readthedocs.io/en/latest/

Hope this helps... if I were you, I'd just get kids to copy files like you, it's certainly the path of least resistance in this case.

Best of luck and please keep the feedback coming!

N.

armbennett, 2016-12-09:
Thanks @ntoll for outlining the options so clearly. I will stick with copying files manually for now as you suggest.

carlosperate, 2017-11-03:
Looks like the original question was already answered, please feel free to reopen otherwise.

closed 2017-11-03 by carlosperate