mu-editor/mu #200: Problems saving with multiple tabs, similar file names

blogmywiki, 2016-12-15:
I'm using Mu on OS X 10.10.4 and have had problems losing data when saving programs with similar filenames (not sure if that's relevant).

I have 2 or more tabs open with code and have had problems with Mu crashing on save and re-opening Mu it reverts to files from previous sessions. On re-opening one of the new .py files I find it completely empty.

In the example below file B began 'file b' but then had about 6 more lines of comments that have vanished.

It may also be triggered by using cmd-S shortcut key multiple times in quick succession to save.

[image]

[mulog.txt](https://github.com/mu-editor/mu/files/654184/mulog.txt)

blogmywiki, 2016-12-15:
I also forgot to add that twice I've had two similarly-named files open in tabs and at some point one file has assumed the title of the other in the tab title. I've not been able to reproduce this and capture a log yet.

blogmywiki, 2016-12-15:
Just had another crash - this time only ONE tab open. Pasted code in from another tab, closed it, saved new tab with new file name. Edited code, some copying and pasting, noticed some odd behaviour as I highlighted with trackpad on MacBook (unexpected scrolling) but all seemed ok. Saved on each edit. Mu then crashed, and newly-saved Python file was saved as a 0-byte file, all text and edits lost. Log file attached. On reopening Mu it defaulted to files I had open in session before last, not the one I was working on when it crashed. Very odd! Mu now pretty unusable on my machine I have to say!
[mulog.txt](https://github.com/mu-editor/mu/files/654648/mulog.txt)

carlosperate, 2016-12-15:
Had a quick look at the log, as you say it seems to be crashing on file write. At least you'll be able to recover the file contents from the log file.
It would be quite helpful if you could start executing mu from the terminal with `./mu.app/Contents/MacOS/mu` hopefully that way any uncaught exception should still be printed there and we could see what might be the problem.
Thanks for persevering! we definitely want to get to the root cause of this.

carlosperate, 2016-12-15:
Btw, did the weird scrolling happen when selecting a word on the editor? I think I might have introduced a bug in that area, I'll look into that once we can get this sorted.

blogmywiki, 2016-12-15:
Yes scrolling happened when selecting words/lines I think. Will try launching Mu from command line.

blogmywiki, 2016-12-15:
This also is probably important - when editing files prior to crash the asterisk did NOT appear at the end of the filename in tab to show unsaved edits.

ntoll, 2016-12-15:
@blogmywiki I'm not on the correct operating system to test this (I use Linux). However, I can play the role of Sherlock Holmes :mag: in the hope something I say may suggest something to you.

Put simply - I'm anxious to prove it's not to do with *your* computer rather than just Mu itself. 

Why?

Many Mu users are Mac users and this is the first report of this kind. If Mu were so fundamentally broken we should be seeing several messages of this kind from OSX users. I'm not saying there *isn't* a problem with Mu, nor that it's *your fault* (heaven forbid!) ~ rather, it strikes me as "fishy" that only you appear to have this problem. Could there be something unique with your set-up about which we don't know? Thoughts? It might be something as random as the amount of disk space you have left, memory size and usage by Mu, other software running at the same time as Mu (that may interfere with files that Mu cares about) and so on. Note! These are all guesses, suggestions and hunches. You know your system, so what 

The code at which Mu crashes is simply Python writing a file to your file system. This is one aspect of the application that interacts with the hosting operating system. That it sometimes works and sometimes crashes (using exactly the same line of code in Mu) suggests that something other than Mu (which remains the same) is changing. What's this other thing in this context? Your operating system.

I wonder if we should add something like https://github.com/ColinDuquesnoy/qcrash into Mu for this sort of circumstance..?

Is this helpful? Thoughts..?

blogmywiki, 2016-12-16:
@ntoll Yes I feel it is very likely something specific to my computer. I noticed that when I started Mu from the command line I got a clean start as if it had never been run before and so far not had any more issues. I have 26GB free on my hard disk so not sure space is the issue. Permissions issue perhaps?

ntoll, 2016-12-16:
Could be. I think this ticket is very helpful in so far as it highlights the need for us to get getter at logging things when stuff goes wrong.

ntoll, 2018-06-04:
Closing since this can't be reproduced. BTW -- finding the logs is now a lot easier.

closed 2018-06-04 by ntoll