mu-editor/mu #201: Unable to launch version 0.9.13 on Mac

jrmhaig, 2016-12-15:
I have downloaded the Mac package for version 0.9.13 from http://codewith.mu/ but it will not open. Running from the command line I get:

```bash
$ /Applications/mu\ 3.app/Contents/MacOS/mu 
Error loading Python lib '/var/folders/6q/v8c0hl0n1p1cy8jfzcwzqdt8tzd1h2/T/_MEIKwrhy1/Python': dlopen(/var/folders/6q/v8c0hl0n1p1cy8jfzcwzqdt8tzd1h2/T/_MEIKwrhy1/Python, 10): Symbol not found: _faccessat
  Referenced from: /var/folders/6q/v8c0hl0n1p1cy8jfzcwzqdt8tzd1h2/T/_MEIKwrhy1/Python
  Expected in: /usr/lib/libSystem.B.dylib
 in /var/folders/6q/v8c0hl0n1p1cy8jfzcwzqdt8tzd1h2/T/_MEIKwrhy1/Python
```

carlosperate, 2016-12-15:
Hi Joe,

You've mention on the slack channel that you were trying to run this in 10.9, is that correct?

It looks like we have a similar limitation in OS X than we do in Linux with only forward support from the build environment OS version.
We currently use Travis CI to create the executables, and they are dropping the 10.9 images on the 20th of January (they've already dropped support in October), so it looks we might only be able to support 10.10 or higher. At the moment is using the default image, which is 10.11, a few months ago it was 10.9, probably why older versions worked on your environment. I've created a new build with 10.10, could you try that one in case it works?
https://s3-us-west-2.amazonaws.com/ardublockly-builds/microbit/osx/mu-2016-12-15_11_27_42.zip

jrmhaig, 2016-12-15:
Yes, I am on 10.9.5.

With the new build I am still getting:

```bash
$ /Applications/mu\ 2.app/Contents/MacOS/mu 
Error loading Python lib '/var/folders/6q/v8c0hl0n1p1cy8jfzcwzqdt8tzd1h2/T/_MEIljRPlC/Python': dlopen(/var/folders/6q/v8c0hl0n1p1cy8jfzcwzqdt8tzd1h2/T/_MEIljRPlC/Python, 10): Symbol not found: _faccessat
  Referenced from: /var/folders/6q/v8c0hl0n1p1cy8jfzcwzqdt8tzd1h2/T/_MEIljRPlC/Python
  Expected in: /usr/lib/libSystem.B.dylib
 in /var/folders/6q/v8c0hl0n1p1cy8jfzcwzqdt8tzd1h2/T/_MEIljRPlC/Python
```

carlosperate, 2016-12-15:
Okay, even if we have to change it in a month, I've configure it for 10.9.5. Could you try this one? 
https://s3-us-west-2.amazonaws.com/ardublockly-builds/microbit/osx/mu-2016-12-15_13_16_05.zip

jrmhaig, 2016-12-16:
Yes, that one works with 10.9.5.

PR #205 (2017-01-17): Select older OS X version for Travis build
With Pyinstaller on OS X we are only guaranteed compatibility on the OS version used for the build or newer. Travis was building on the default image 10.11 at the moment, so this PR reduces this to the lowest available, 10.9.5. Unfortunately this version is no longer supported and will be removed from Travis in January 20th, but we should use it for as long as we can and include this information in the release notes. 

Fixes #201
closed 2017-01-17 by ntoll