mu-editor/mu #202: Outdated setup instructions for Mac

jrmhaig, 2016-12-15:
The instructions for Mac in `README.md` and `package/README.md` say that PyQt5 and QScintilla2 should be installed using brew. However, these can now be installed using pip. In particular, the instructions for QScintilla2 will not work as the file `extras/qscintilla2.rb` is missing.

inglesp, 2017-01-22:
`extras/qscintilla2.rb` was removed in 5b098ca9789ce539d039277a6633c22a0669b991, where @carlosperate says "(PyQt & QScintilla will be absorbed in requirements.txt once ready)".  @carlosperate, when will this be?  I can do the work to tidy up the documentation.

So that it doesn't get lost, I was able to set up a development environment on OSX with the following:

```
env PYTHON_CONFIGURE_OPTS="--enable-framework" pyenv install 3.5.2
pyenv virtualenv 3.5.2 mu
pyenv activate mu
pip install -r requirements.txt 
pip install PyQt5
pip install QScintilla
pip install git+git://github.com/pyinstaller/pyinstaller.git@483c819d6a256b58db6740696a901bd41c313f0c
pyinstaller package/pyinstaller.spec
```

carlosperate, 2017-01-22:
Well, the entire thing with installing and using PyQt the same way in all platforms has been a bit of a headache for the past 6 months or so.

Ignoring the webview functionality we'll have to use in the near future for the blocks integration (and figuring out if we can still use the same implementation, and therefore the same PyQt version, in all platforms will be it's own headache soon enough), this is how we currently stand:

* Windows/Linux/macOS all are finally good to install all dependencies with pip as far as the users are concerned (we had an issue with Qscintilla in Linux, which was fixed during christmas) ✔️ 
* Travis Linux and macOS server builds currently use pip to install PyQt and Qscintilla, so we are up-to-date for these two platforms ✔️ 
* AppVeyor for windows has to stick with Python 3.4 because of an issue with Python 3.5 and PyInstaller, therefore we cannot install PyQt with pip ❌ 
    * Technically speaking the same Christmas released included a wheel version of PyQt and Qscintilla for Python 3.4, but one of their dependencies (SIP) has only been released for 3.5
* https://github.com/mu-editor/mu/issues/216: We need to add `install_requires` to the setup.py file. ❌ 
* https://github.com/mu-editor/mu/issues/217: We need to pick and implement a way to separate "user dependencies" from "development dependencies". There is many ways to do this, it's mostly a matter of picking one and submit a PR. :x:
* requirements.txt file does not yet contain the PyQt/Qscintilla dependencies: :x:
    * We could include it and not use this file in AppVeyor
    * We could not update it  yet and wait for SIP to be ported to Python 3.4
    * We could have different things in the requirements.txt file and some type of platform-specific install requirements in setup.py
* Installation documentation needs to be updated (this depends on how requirements.txt and setup.py have been configured from the points above) :x:

JaxonKneipp, 2017-02-05:
Me and some friends are really keen on doing some work on Mu. However I am running into problems in regards to the setup process as outlined above. Particularly in regards to the installation of Qt5 within the project. Is there a place now where we can follow up to date setup instructions designed for people new in this field?

carlosperate, 2018-03-17:
The latest beta version installs all dependencies from PyPI, so we have more or less the same instructions for all platforms when running from source.
I'll close this one as I would say the original issue has already been addressed, but please feel free to reopen it if something isn't clear or doesn't quite work as expected. Thanks!

closed 2018-03-17 by carlosperate