mu-editor/mu #206: OS X build is unsigned, downloaded from a plaintext website

glyph, 2016-12-16:
The "getting started" process for Mu teaches new programmers on macOS a _terrible_ security habit:

1. https://codewith.mu has an invalid certificate
2. the application bundle is unsigned, meaning it won't launch if you've properly secured your mac

This leaves Mu in [venerable company](https://glyph.twistedmatrix.com/2015/11/editor-malware.html), but both of these issues really ought to be fixed.  Teaching people that there are serious consequences to plaintext downloads is hard enough without useful projects providing no way to securely get their software.

ntoll, 2016-12-17:
Morning @glyph - we're painfully aware of these shortcomings. We're a young project run by time poor volunteers. That's not meant to be an excuse, but it at least demonstrates why we've not done these things yet.

They're on our list and we're getting there.  Thanks for letting us know!

ntoll, 2016-12-17:
For reference - details on adding TLS to custom domain hosted on GitHub pages: https://gist.github.com/cvan/8630f847f579f90e0c014dc5199c337b

ntoll, 2016-12-17:
@carlosperate anything to add on signing releases?

carlosperate, 2016-12-17:
Indeed, this is something that's on the to-do list. The other thing about signing the applications is that there is a cost to get a Windows certificate and to become an Apple Developer.
Wayne was just telling me about CloudFlare for github pages last Thursday, @WayneKeenan were the instructions that @ntoll linked what you were referring to?

Also, for reference there are individual issues already opened for these topics: https://github.com/mu-editor/mu/issues/148 https://github.com/mu-editor/mu/issues/177 https://github.com/mu-editor/mu/issues/194.

glyph, 2016-12-17:
Perhaps the PSF would be interested in sponsoring those certificates?

TLS, at least, should be free via let's encrypt.

glyph, 2016-12-17:
Oh hey I see @ntoll already suggested that on the relevant issue :).  If this is just a duplicate of existing stuff, I'm happy to see it closed.

ntoll, 2016-12-20:
OK... I've just kicked off migration to CloudFlare - which will give us free TLS support.

ZanderBrown, 2016-12-20:
Any way to force HTTPS? https://codewith.mu works fine but just entering codewith.mu still uses HTTP

ntoll, 2016-12-21:
Yup... just wanted to ensure TLS (eventually) worked. Didn't want to put the site into a bad state by default. I'll pull the lever / press the button / cast the enchantment right now. ;-)

WayneKeenan, 2016-12-21:
@carlosperate yep, that's it.

ntoll, 2016-12-21:
OK - lever has been pulled. We serve over TLS by default and redirect to a secure endpoint if the user uses just `http://codewith.mu/`

carlosperate, 2016-12-21:
That's great! Can confirm it works as expected :D
We can now close https://github.com/mu-editor/mu/issues/194 as well.

And thanks for confirming @WayneKeenan (might end up doing the same for my domain, seeing that https is needed for web-bluetooth pages).

glyph, 2016-12-22:
Thanks a bunch for making this happen.  Security educators everywhere salute you for making their jobs a little easier :)

glyph, 2016-12-22:
I'm going to close this now, since the download is HTTPS, at least, so there's _some_ authenticity. Codesigning would be nice but at least users are not operating on zero assurance.

closed 2016-12-22 by glyph