mu-editor/mu #207: Need to check for file permission before attempting to save.

carlosperate, 2016-12-17:
If we open a file that is read only for the user and try to save we don't safeguard for `PermissionError`:

```
Traceback (most recent call last):
  File "C:\workspace\git\microbit\mu_original\mu\logic.py", line 601, in save
    with open(tab.path, 'w', newline='') as f:
PermissionError: [Errno 13] Permission denied: 'C:/Users/xxx/mu_code/test_locked.py'
```
We should catch this, and then either:
a) Ask the user to manually remove the permission and try again
b) Ask if the user wants Mu to change the read permission and continue saving

carlosperate, 2018-03-17:
@ntolls this one is still a bug, and now with the autosave it means that Mu crashes a few seconds after opening the read-only file:
```
Traceback (most recent call last):
  File "C:\workspace\mu-temp\mu\mu\logic.py", line 879, in autosave
    with open(tab.path, 'w', newline='') as f:
PermissionError: [Errno 13] Permission denied: 'C:/Users/carlo/mu_code/hello.py'
```

tjguk, 2018-03-17:
I'll pick this up for now, since I'm working in the load/save space for issue #370. At least, please can others leave this area alone for the moment?

carlosperate, 2018-03-17:
Thanks Tim!

ntoll, 2018-03-20:
+1000 @tjguk

ntoll, 2018-03-21:
Fixed by #399. I added a helpful message in a948debc2fe769df591effd97b1dd523f59beecc.

closed 2018-03-21 by ntoll