mu-editor/mu #211: REPL windows does not work if you swap microbits

gingemonster, 2016-12-20:
If you have two microbits, mu and windows, and do the following:

1. Plug one in to usb and open my
2. flash some code to the microbit and open REPL window
3. Uplug the microbit and plug a different microbit in
4. Save code to the new microbit (successfully)
5. Click REPL button to close it, and again to re-open

It will bring up an error saying no microbits are connected, even though it happily just flashed code to the new microbit.

I suspect its locked to one microbit per application session.

Here are the relevant lines from the log starting from point 4 above:

mu.logic(flash) DEBUG: Path to micro:bit: F:\
mu.logic(flash) DEBUG: Flashing to device.
mu.interface(show_message) DEBUG: Flashing "main.py" onto the micro:bit.
mu.interface(show_message) DEBUG: When the yellow LED stops flashing the device will restart and your script will run. If there is an error, you'll see a helpful message scroll across the device's display.
mu.logic(add_repl) INFO: Starting REPL in UI.
mu.logic(find_microbit) WARNING: Could not find micro:bit.
mu.logic(find_microbit) DEBUG: Available ports:
mu.logic(find_microbit) DEBUG: []
mu.interface(show_message) DEBUG: Could not find an attached BBC micro:bit.
mu.interface(show_message) DEBUG: Please make sure the device is plugged into this computer.

The device must have MicroPython flashed onto it before the REPL will work.

Finally, press the device's reset button and wait a few seconds before trying again.

ntoll, 2016-12-20:
This depends upon the host OS working in a timely fashion. :-(

Sometimes it can take 30+ seconds before the device appears as an available port to which you can connect the REPL. 

Essentially, under the hood PySerial is probing for available USB serial ports and Mu uses the reported PID/VID combinations to work out to which port the device is connected in order to connect to the REPL running on the device. This usually doesn't take any time at all. 

However, *sometimes* it takes time for the port to be correctly reported by the OS. My only advice is to wait a bit and try again (this is something beyond Mu's control to fix).

Sorry we can't be more helpful in this instance. :-(

closed 2016-12-20 by ntoll

carlosperate, 2016-12-20:
@ntoll should we add a time stamp to the log? Would help to identify how long between event

ntoll, 2016-12-20:
Good catch. See 40263950bdb7430dd345de01c62e035309c8d372