mu-editor/mu #212: Outline spec for highlighting selection

tjguk, 2016-12-21:
We recently added functionality to highlight text which matches any text currently selected. Issue #209 highlighted an issue when selection was made by keyboard. In the course of reworking the code and tests to fix that, the following questions appear to be implied but not specifically answered by the existing implementation. My own feelings are given in square brackets.

1) If *less* than one word is selected, should all matches be highlighted? This would include a single letter.

[Yes, but with a limit per (2) below]

2) If less than one word is to be matched and highlighted, should there be some lower bound, eg, only once more than 3 letters have been selected?

[Yes, only once 3 characters have been selected]

3) Should the selected text -- which obviously matches itself -- be highlighted? At present it is, but there doesn't seem to be much sense to this: when selected, the highlight is hidden; when selection is changed, the highlights will be cleared? Although cf (4) below

[No; don't highlight selected text]

4) Should highlights be cleared once the selection is cleared (not just changed)? If not, then it makes sense to highlight the selection itself as in (3) above. But then it's not clear how to *un*highlight without artificially adding another selection.

[Yes; clear highlights on unselection]

ntoll, 2016-12-21:
@tjguk this is great! I agree with each of your conclusions for each numbered comment. It was @carlosperate who originated this feature IIRC and I believe the intention was simply to allow users to quickly see where related things are in the code. Nothing more. I'd very much err on the side of simple=good. :-)

carlosperate, 2016-12-21:
Thanks @tjguk for looking into this, when I worked on the original implementation I was hoping to only use the default Qscintilla API as to not reinvent the wheel. However, as you've pointed out, it oddly traverses through the code changing the selections, so this should definitely be re factored as you have been doing.
The idea was to mimic the functionality available in a lot of the popular GUI editors (so thinking along the lines of default behaviour in things like sublime, notepad++, or even the eclipse editor).

> 1.  If less than one word is selected, should all matches be highlighted? This would include a single letter.
>
> [Yes, but with a limit per (2) below]
>
> 2. If less than one word is to be matched and [...]

Unless this complicates the code too much I would argue the opposite here. Most editors I've used will do the highlight only when a single word is selected, following the same rules that are applied to how much is selected when you double click on a character. And then they will only highlight the same word matches (well, some will automatically select the word where the cursor is placed, and then do the same).

Obviously if a lower limit is not applied, then it becomes a bit chaotic when the selection starts, highlighting half your script, changing as you selection changes. Apart from that I was slightly concerned on performance (we want to make sure it still runs all right on old raspberry pies), since the event will be fired on every selection change and users could be doing very large mouse drags very quickly.

This would also be because normally you want to highlight varaibles, functions, and similar, so the "is it a full word" rules apply really well here. If the user wants to find all matches for a specific string we should be introducing a find box (which should reuse all the code written for this implementation)

> 3. Should the selected text -- which obviously matches itself -- be highlighted? At present it is, but there doesn't seem to be much sense to this: when selected, the highlight is hidden; when [...]

I think at this point we need to distinguish between text selection highlight and search results highlight. When we select text it will automatically highlight it in blue, and any matched results will be highlighted in grey.
I can't remember without looking at the implementation, but I assume you are asking this because the grey highlight is being applied "below" the blue on on the selected word. Personally I don't think it matters that much, I would just decide this on whatever is the easiest/simplest to implement.

>  4.  Should highlights be cleared once the selection is cleared (not just changed)? If not, then it makes sense to highlight the selection itself as in (3) above. But then it's not clear how to unhighlight without artificially adding another selection.
>
> [Yes; clear highlights on unselection]

I agree with this point, some editors leave it, but I don't see much value in doing so since this is the type of editor where users are not likely to navigate the code with the keyboard only. I'd rather remove the visual clutter, and I do agree with you that if we do leave it then it is not blatantly obvious how to unselect matches.

tjguk, 2016-12-21:
Thanks for the considered feedback, @carlosperate; exactly what I was hoping for. 

1 & 2) Highlight incomplete words? Funnily enough, in the absence of anything else, my preference is the same as yours. However, [your] existing implementation clearly allowed incomplete words and I was trying for what I thought was a compromise! I know @ntoll had previously agreed with my compromise, but unless he has strong feelings, I'm very happy to implement complete-only here. 

3) Highlight the selection? In fact, having just tried it out: the "indicator" highlight overrides the "selection" highlight. I'm actually +0.5 on *not* highlighting the selection as I think it could be confusing one way or the other. Implementation is trivial: I've actually removed the two lines it took, but I can put them back in!

4) We seem to be agreed on and, IIRC, this is what the current implementation does anyway.

carlosperate, 2016-12-21:
> However, [your] existing implementation clearly allowed incomplete words and I was trying for what I thought was a compromise!

(Answering from memory at the moment, since I cannot really look at the code now) I think the "text selection match" was configured to do full-word matching. However the the function that finds the matches should have an optional argument to select between full word or not. I left that available with the prospect of reusing all of this for a general search box, which I think we should probably add to the roadmap as well (a simple box that appears with a key binding, like Ctrl+F).

> 3. Highlight the selection? In fact, having just tried it out: the "indicator" highlight overrides the "selection" highlight. I'm actually +0.5 on not highlighting the selection as I think it could [...]

The original implementation should have ultimately shown the "selected word" in blue, and all "indicator matches" in grey. I remember making sure that was the case when I was tinkering with it, as it is important to visually separate the two, but there wasn't anything specific on the code to ensure one was on top of the other. If it's a trivial change I'd agree with you and say that it's better for the "grey highlight" not to be applied to the the selected text at all.

tjguk, 2016-12-21:
On 21/12/2016 12:55, Carlos wrote:
> (Answering from memory at the moment, since I cannot really look at the
> code now) I think the "text selection match" was configured to do
> full-word matching.

Ok; I see. In fact the complete-word check is a regex [*] which will 
correctly exclude more than one word but, unless it peeks *outside* the 
selected text, will not know whether the selected text matches word 
boundaries or not.

[*] '^[A-Za-z0-9_-]*$'

ntoll, 2016-12-21:
For the record - I defer to your collective decision. :-)

ntoll, 2018-06-04:
Closing since the related PRs have been merged.

closed 2018-06-04 by ntoll