mu-editor/mu #214: Warning, badly formatted line on  mu/debian/changelog

carlosperate, 2016-12-21:
When trying to build a .deb file I get the following warning:

```
...
parsechangelog/debian: warning: mu-0.9.13/debian/changelog(l5): badly formatted trailer line
LINE:  -- Nicholas Tollervey <ntoll@ntoll.org> Sun, 06 Oct 2016 11:32:58 +0000
...
```

Sounds a bit silly, but it might be that it needs a double white space before weekday.
I'm not quite sure when I'll be able to give this a try, but it is here for tracking and in case anybody else wants to give it a go.

ntoll, 2016-12-21:
What's the command to make this happen?

carlosperate, 2016-12-21:
You can see the build script I'm currently using here: https://github.com/carlosperate/mu/blob/pi_ci_scripts/package/pi_ci_scripts/build.sh

ntoll, 2016-12-21:
aha! I knew it'd be somewhere.

carlosperate, 2016-12-22:
Can confirm the warning does not appear any more with https://github.com/mu-editor/mu/commit/607bbb398add9072fe032a537e98941dd6cb06b4.

closed 2016-12-22 by carlosperate