mu-editor/mu #217: Separate requirements.txt and requirements-dev.txt?

carlosperate, 2016-12-22:
Juts a suggestion, as it seems like a good idea. Most uses do not need to install the packages required for testing.

ukBaz, 2016-12-22:
Does this help you at all? 
http://setuptools.readthedocs.io/en/latest/setuptools.html#declaring-dependencies

There appears to be a extras_require feature

carlosperate, 2016-12-22:
Hi Barry,
Yes, good point, we should also separate it on the setup.py configuration, but without having two requirement files we would have to manually list the same packages into both the requirements and setup.py files. Alternatively if we have requirements.txt and requirements-dev.txt, then setup.py can read these, and there is no risk for them to desynchronise.

WayneKeenan, 2016-12-22:
checkout the september 2016 update of:
http://stackoverflow.com/questions/28509965/setuptools-development-requirements

All the best
Wayne

carlosperate, 2016-12-22:
Thanks for the link @WayneKeenan, definitely good to know, as this is an area I didn't know much about.

Based on that info and the idea to still separate "install only" and "dev" type of dependencies I think I would still prefer the `requirements.txt` and `setup(install_requires=[...])` to be pointing to exactly the same list (with or without explicit package versions, as needed), and it only being explicitly declared in a single location (the requirements.txt or the setup.py file).
We could then either add the "dev only" dependencies to the `setup(iextras_require={'dev': [...]})` and install them using `pip install -e .[dev]` (which I personally don't really find obvious, but it should be fine as long as it's documented), or to have them in a `requirements-dev.txt` file and read that inside the `setup.py` file to be able to also use `pip install -r requirements-dev.txt`.

@ntoll, any thoughts?

ntoll, 2016-12-23:
Morning!

Ah, Python packaging (there must be one, and only one obvious way to do... oh hang on a minute...) ;-)

I guess it's simply a case of choosing one thing and sticking to it. I'd look for a solution that is both as simple as possible and obvious for "install only" - to me that would be `pip install mu`. If you need a development environment I guess remaining dependencies are explicitly found somewhere (to my mind, that's usually a `requirements-dev.txt` file). @carlosperate I like your second suggestion of reading such a file in `setup.py` so developers can do either `pip install -r requirements-dev.txt` or `python setup.py develop`.

Comments, constructive critique and ideas most welcome!

Another thing we *must* do is pin the versions for dependencies on each release.

ntoll, 2018-06-04:
Closing since this is addressed in developer docs and the new version of the website (coming soon).

closed 2018-06-04 by ntoll

mcandre, 2018-10-16:
Lol, "requirements-dev.txt" is enough of a convention that someone published https://pypi.org/project/requirements-dev.txt/ , to retrofit a safeguard against people accidentally running `pip install [no -r] requirements-dev.txt`, just like pip does with the standard `requirements.txt`