mu-editor/mu #223: How to import self-written Python scripts?

jessiejc23, 2016-12-31:
Is it possible to import self-written Python scripts on Mu editor? I wrote one for which I need it to connect the servo motors to my micro:bit.

jezdean, 2016-12-31:
There's a guide [here about it](https://github.com/microbit-playground/microbit-servo-class/blob/master/README.md).  You copy it over by clicking the `files` button in mu. 

Perhaps info about it should be on a website or in the micropython docs. Happy to add it but not sure where to put it?

jessiejc23, 2017-01-02:
@jezdean maybe consider opening an 'Other applications of micro:bit' section in the micropython docs?

Also, does the script work with having the servo motors connecting to a PWM board on pins 0-3 (which connects to a micro:bit with an edge connector)?

jessiejc23, 2017-01-02:
Further research got me this: https://github.com/adafruit/Adafruit_Python_PCA9685 will test how to make it work.

jezdean, 2017-01-02:
I've not tried to use one of those boards I'm afraid. The driver you linked to is for standard micropython and will not work out the box on the microbit; you'd need to make changes.

But you do not need a PWM board to drive a servo. The microbit as-is will just drive it from pin0.

Connect the three servo wires to the microbit's pins:

| Servo wire | Microbit pin |
---------|--------------
| Red (or brown) | 3V |
| Black | GND |
| Orange (usually labelled SIG or PWM) | pin0 |

PWM is strange on the microbit when using `mu` at the moment. The servo will work but it will be jittery. Consider using the microbit.co.uk online python editor until it's fixed.

jessiejc23, 2017-01-03:
Thanks for your explanation. :) But I need to use one micro:bit to control/connect 4 servo motors, hence I needed a PWM board. I will try it on the online python editor and update here if there is any positive progress.

jessiejc23, 2017-01-04:
@jezdean I am now using my microbit to connect 2 servo motors (without the pwm board). It only seems to working stably if only one servo is connected to my micro:bit. Not sure if you have encountered the same issue before? Is it due to insufficient power/battery?

carlosperate, 2017-11-03:
As it has been almost year I assume your original question as already been answered (we also have https://github.com/mu-editor/mu/issues/224 and https://github.com/mu-editor/mu/issues/259). If not, please feel free to reopen this issue.

closed 2017-11-03 by carlosperate