mu-editor/mu #224: Importing Python modules on Mu/online python editor

jessiejc23, 2017-01-03:
I need to use this Python script https://github.com/adafruit/Adafruit_Python_PCA9685/blob/master/Adafruit_PCA9685/PCA9685.py on mu editor but the 4 Python modules (division, time, math, logging) in this script do not appear to have been included in this editor. 

Is it already possible to import these modules manually on Mu? If so, how? If not, would the online Python Micro:bit editor do the job?

gingemonster, 2017-01-29:
Hi

That library would not work on the microbit without modification, luckily that's what I did yesterday https://github.com/gingemonster/PCA9685-Python-Microbit

There are links on the projects readme page for how to "load" other script files in mu

carlosperate, 2017-11-03:
The "files" feature should allow adding files into the micro:bit micropython file system. We have issue https://github.com/mu-editor/mu/issues/259 to see if there is something we can simplify and also as a reminder that there needs to be some documentation to properly indicate how to do this.

Since this issue is almost a year old I'll close it as hopefully your questions was either been answered or the discussion could be continued in the previously linked issue. But please, if you have more questions relevant to this specific ticket please do reopen it.

closed 2017-11-03 by carlosperate