mu-editor/mu #225: Delete key broken in REPL

hickford, 2017-01-06:
Hi. I'm using Mu 0.9.13 on Windows. I've noticed that in the REPL, the delete key does not work as expected. Instead of erasing the character to the right of the cursor, it erases the character to the left—as if you'd pressed backspace.

In the code editor, Delete and Backspace work as normal.

ladyada, 2017-08-15:
this seems to be fixed in 1.0.0.alpha.2 - delete is working as expected for me

Sembazuru, 2018-04-16:
Seems be broken on 1.0.0.beta.15 on Win10x64. At least for me on 2 different computers.

Sembazuru, 2018-04-16:
I have Mu configured to start in Adafruit CircuitPython mode, I really haven't used any other modes. 

In the code editor the DELETE and BACKSPACE keys function as expected.

I then click the REPL button to open the REPL/Serial pane. I then press any key to "enter the REPL". I execute a command (for example `help("modules")`). Then I press up arrow to get the previous command and use the left arrow to move the cursor into the previous command. This is where both DELETE and BACKSPACE both function like BACKSPACE.

If the REPL is running internally on the circuit python device, the issue might actually be with the device firmware instead of Mu. Or, maybe the wrong keycode is being sent for the DELETE key? Here is the version string for the installed circuit python on my device:
Adafruit CircuitPython 2.2.4 on 2018-03-07; Adafruit Gemma M0 with samd21e18

ntoll, 2018-04-17:
@Sembazuru OK... you've inadvertently helped to find two bugs (the one in the Python 3 mode code runner which I mention above, and now one in MicroPython which you have described above).

Regarding the bug you mention, my best suggestion is to report it upstream here: https://github.com/micropython/micropython

I'll quickly fix the other bug in Mu.

Thanks for taking the time to give such useful feedback!

ntoll, 2018-04-17:
OK... a fix for Python 3 mode is in master and will be in the next release. The behaviour for the MicroPython REPL will change when upstream is fixed.

dhalbert, 2018-04-20:
Tested this in terminal programs other than Mu REPL: Delete will delete forward if it sends 0x1B 0x4B 0x33 0x7E (ESC [ 3 ~). Backspace sends 0x7F.

Tera Term, out of the box, sends 0x08 for Backspace and 0x7F for Delete, so it only deletes backward. If I change a setting in Tera Term, I can get it to send the combinations in the paragraph above.

I don't think there's anything to change in MicroPython or CircuitPython: it supports the first-paragraph combinations properly.

Is Mu is now sending these combinations in all cases?

ntoll, 2018-04-21:
@dhalbert *doh* my bad. I'll fix. :-)

ntoll, 2018-06-04:
Fixed so closing. Thanks for all the contributions folks! :-)

carlosperate, 2018-06-04:
@ntoll This hasn't been fixed for micro:bit mode, and I assume the same would apply for CircuitPython mode. I can send the "del" command using microREPL and works fine, while it doesn't with Mu.

ntoll, 2018-06-04:
OK... @carlosperate @dhalbert -- I couldn't find the code I'd written originally, so had a real sense of deja vu  with dac6c05 ;-) Who knows where the original fix went.

dhalbert, 2018-06-04:
I think you had a dream.

ntoll, 2018-06-04:
That's happened before...

carlosperate, 2018-06-05:
Thanks @ntoll!

commit bc4fc942 (2018-04-17): Add correct handling of delete/backspace to Python runner pane (and tests). Fixes #225 for Python 3 mode.
commit dac6c05c (2018-06-04): Fix MicroPython based DELETE key handling in the REPL. FINALLY fixes #225