mu-editor/mu #226: Installer/running mu - issue on Windows XP SP3

NKT00, 2017-01-07:
Tried to install/run the exe on Windows XP SP3 today, and got this error when I ran the exe downloaded from codewith.mu:

Fatal Error!
run returned -1

I'm currently installing the behemoth that is QT5 - it'll be done in the morning hopefully! Trying to compile it myself night reveal the issue.

Any ideas? (Besides "Update that OS!")

ZanderBrown, 2017-01-07:
I find the idea of compiling a python program interesting 😄

NKT00, 2017-01-07:
I'll try it on Windows 10 tomorrow. Works fine on Windows 7. But as @ntoll pointed out, there are plenty of XP boxes out there, especially in the education sector.

And ok, yes, interpret. But, it didn't work as a binary, so digging into the code and whatever dependencies etc are required I'll do tomorrow. Not sure what this qt5 thing is all about, either - it the source is Python, can't I just download it and run it? Except, of course, Windows and other platforms vary so much in the GUI etc. that this extra abstraction layer is required...

ZanderBrown, 2017-01-07:
Qt5 is the cross platform widget library Mu uses

My last comment was meant in a sarcastic tone :-)

Did mu get as far as creating settings.json and start logging? Take a look in `<USER>\AppData\Local\python\mu`

carlosperate, 2017-01-08:
It won't be realistic to keep supporting XP taking in consideration is unsupported by Microsoft and that all major projects (including Mu dependencies) have stopped supporting it as well.

I remember looking at this a while ago (although I cannot really find anything here so it might have been in the microbit mailing list or on twiter). You would have to install an older version of python and find an older version of PyQt5 compatible with that.

Could you try running this version as well? https://s3-us-west-2.amazonaws.com/ardublockly-builds/microbit/windows/mu-2016-11-08_12_23_50.exe
It should open a terminal window where more information should be printed.

NKT00, 2017-01-11:
Here's the output. Any easy way to hold the command window open?

[image]

carlosperate, 2017-03-03:
Sorry for the late reply @NKT00, I think that if you need to use Mu in Windows XP you'll have to install the latest python version for XP (https://www.python.org/ftp/python/3.4.3), make sure pip is installed with Python, and then find the compatible PyQT5 (http://stackoverflow.com/questions/40008236/latest-versions-of-python-3-and-pyqt5-working-on-windows-xp-32bit).

After that it should hopefully work after the other packages have been pip installed:
```
pip install -r requirements.txt
```
If pip is not available you could navigate to the (default) `C:\Python34\Scripts` directory in the terminal and it should work from there:

```
cd C:\Python34\Scripts
pip install -r <Mu_repository_directory>\requirements.txt
```

carlosperate, 2017-11-03:
I hope you were able to solve your issue @NKT00, since it's been a while I'll close this ticket, feel free to open it again if you still have questions.

closed 2017-11-03 by carlosperate