mu-editor/mu #227: Upgrade firmware inside Mu

DavidWhaleIET, 2017-01-11:
Feature Request:

Is there a way from the GUI of Mu to upgrade the MicroPython firmware that is inside the package? Or is there an easy 'file copy' at the OS level that could be written up in instructions?

We have a number of cases now where the response from the Python team has been 'use the latest firmware, here is a link to where you could rebuilt it yourself'. This is way beyond the level at which a number of our users want to engage. In one case, Damien built a version of the .hex for us to use, but again, this requires the user to install the full version of Python on their machine and engage at the command line with hexlify and command line parameters, something they are not familiar with or comfortable with, so the activity just stalls there and we don't get any testing done by the wider user community, and they loose interest as 'its too hard'.

If Mu had an option in the GUI that said 'use this firmware.hex file' - we could get Mu to do all the heavy lifting for us, and it would be easier for users to upgrade their firmware with test versions when trying to prove if a particular firmware version fixes the problem or not, without having to download other tools and engage at the command line.

This is specifically for issues where the 'latest firmware' has not yet made it into Mu.

P.S. Calling it 'firmware' is a bit of a misnoma actually. It's not firmware, as it is built into the application that is downloaded every time a new application is downloaded. Firmware is usually reserved for 'software that does not change much that is pre-programmed into a device'. It's a minor point, but it causes confusion with users.

The only firmware on the micro:bit is the firmware that is pre-programmed into the USB interface chip and managed with MAINTENANCE mode. The software you are referring to here is actually runtime code and the MicroPython interpreter.

ntoll, 2017-01-11:
1. "It's too hard" is an educational opportunity @DavidWhaleIET. I will move mountains to write instructions, help answer questions and talk people through the steps they need in order to accomplish things with the command line. If you don't feel you can provide this level of support just pass them over to me and I'll help. "It's too hard" is the enemy of progress. "It's too hard" is giving up. "It's too hard" is never the right answer. :-)

2. "Firmware"... that's what it's always been called in the MicroPython world. That's what the build tools name it. That's what we have in our docs. Strangely enough, *we're well aware* that it's the runtime code and MicroPython interpreter. This feels like petty semantics: I can find plenty of definitions of "firmware" that make our use legitimate (see Wikipedia for an example). You say Tom-ay-to, I say Tom-ah-to... etc... :-)

3. I don't think your suggestion for Mu to upgrade firmware is sensible for a number of reasons:
  * Mu is meant to be simple. We work *very hard* to keep it simple. For this reason, adding new complicated features that could be addressed in other ways is not the way to go (e.g. just update to the latest version of Mu).
  * Your suggestion also opens the Pandora's box of combinatorial problems of multiple versions of Mu working with multiple versions of the firmware. This is a debugging nightmare waiting to happen.
  * Where do you suggest the latest firmware be downloaded from? How should this be checked? What about mitigation of MiM attacks? As a user, how do I obtain the permissions to update the firmware (if the editor is installed for multiple users)? Do I update it just for me or for all users of my system? Who pays for infrastructure? Who maintains the infrastructure? Who has the role / responsibility for release? We're a bunch of volunteers who work on this when we have the time. What you suggest feels like it's going to cause an unsustainable and complicated time-sink for these volunteers who are already over-stretched as it is.
  * We already have the problems you stated in hand. These will be fixed by the latest version of Mu when it's released, *when it's ready*.
  * Mu is moving to becoming a generic beginner's editor for Python. Adding such micro:bit specific functionality would simply add more complexity at a time when the editor is undergoing a transition that we must get right.

I'm not saying it'll never happen. ;-) I have a "rule of three" for new features. If three different people who don't know each other suggest a feature independently of each other I see this as a good indication that such a thing ought to be addressed. For example, with the Python blocks that'll be landing in the next version, Jonny, Adafruit and Sway all mentioned such a capability independently of each other.

Does this make sense? Happy to answer questions, move mountains and continue to play devil's advocate.

DavidWhaleIET, 2017-01-11:
The user in question has already asked the python community for help with
trying to install the latest firmware, so the door is open, please help him
ta :)

ntoll, 2017-01-11:
@DavidWhaleIET do you have a link or something to point me in the right direction? I get 100s of emails a day and I'm likely to have missed this.

DavidWhaleIET, 2017-02-01:
This issue was dealt with via email. The solution is as below:

The idea came from Nicholas late last night!

1. go to: https://github.com/bbcmicrobit/PythonEditor

2. press the CLONE OR DOWNLOAD button

3. Choose the DOWNLOAD ZIP option

This will download a complete web based python editor.

4. Unzip this on your computer - you should now have a PythonEditor folder

5. get Damien's patched .hex file that we sent you

6. rename Damien's .hex file as 'firmware.hex' and replace the firmware.hex that is in the PythonEditor directory that is now on your computer with this patched version.

7.  double click on editor.html - this should launch a complete python editor web page in your browser.

8. Paste your code into this editor window, then press the DOWNLOAD button. This will generate a .hex file that has your script and Damien's latest MicroPython 'firmware' inside it (which apparently has the PWM fix inside it that he was talking about)

9. Test your code again to see if the PWM lasts for longer, and report back to us on how this goes?

closed 2017-02-01 by DavidWhaleIET

fizban99, 2017-02-01:
Does this also fix a https://github.com/bbcmicrobit/micropython/issues/335 ? 
Is this patched .hex file available somewhere?

whaleygeek, 2017-02-01:
@ntoll ??

ntoll, 2017-02-01:
OK... I just quickly checked and it all seems to work properly (to my musician's ears) no matter which one of speech or music you use first.

fizban99, 2017-02-02:
Good to know. Thanks!. Is there an estimate for when this firmware will be incorporated into Mu?
Is it possible to get the patched .hex from somewhere as a workaround?