mu-editor/mu #231: MU5 crash when displaying a DEGREE sysmbol

bobh46320, 2017-01-19:


ntoll, 2017-01-19:
Hi,

Thanks for reporting this.

Unfortunately, I cannot reproduce the problem. I assume you mean the "°" symbol..? I'm able to display the symbol in Mu, save and load files containing the symbol and flash code containing the symbol onto a micro:bit.

I'll need a bit more information to help me work out the problem:

* What operating system are you using?
* Can you attach the log file?
* Do you know which version of Mu you're using (are you using the latest version?)?
* What steps do I need to follow to reproduce the problem?

Thanks! :-)

bobh46320, 2017-01-19:
from microbit import *
display.scroll("°")

The above code will crash MicroPython for a MicroBit.  At the time of the crash, it will delete all source code from the source program file and then crash MU. I appreciate that the degree sign can't be written to the MicroBit LED array,  but I am outputting 4 bit strings to a 16 x 2 LCD screen where the symbol is legitimate.
I am running MU5 on a MAC.
Mac Stack Dump:-
Path:                  /Users/USER/Documents/*/mu-5.app/Contents/MacOS/mu
Identifier:            mu
Version:               0.0.0 (???)
Code Type:             X86-64 (Native)
Parent Process:        ??? [1]
Responsible:           mu [25223]
User ID:               501

Date/Time:             2017-01-19 13:37:09.610 +0100
OS Version:            Mac OS X 10.12.2 (16C67)
Report Version:        12
Anonymous UUID:        D2E21157-AF7F-3AD2-B491-B1E9C6114EA2

Sleep/Wake UUID:       21634F2E-47A4-4978-B97E-AA62CC709E7A

Time Awake Since Boot: 150000 seconds
Time Since Wake:       210 seconds

System Integrity Protection: enabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_CRASH (SIGABRT)
Exception Codes:       0x0000000000000000, 0x0000000000000000
Exception Note:        EXC_CORPSE_NOTIFY

Thread 0 Crashed:: Dispatch queue: com.apple.main-thread
0   libsystem_kernel.dylib        	0x00007fff95e5cdd6 __pthread_kill + 10
1   libsystem_pthread.dylib       	0x00007fff95f48787 pthread_kill + 90
2   libsystem_c.dylib             	0x00007fff95d73497 raise + 26
3   mu                            	0x0000000100004e47 0x100000000 + 20039
4   mu                            	0x0000000100002dbc 0x100000000 + 11708
5   mu                            	0x0000000100001804 0x100000000 + 6148

ntoll, 2017-01-19:
Hi @bobh46320,

It works for me with the latest version of Mu. Given that there's no symbol for "°" built into MicroPython I see the expected "??" scroll across the screen.

ntoll, 2017-01-19:
Aha... I see you just closed the ticket. I assume the latest version fixes things for you.

bobh46320, 2017-01-19:
Just loaded latest MU dated 18/01/2017 - same error.  Is there a difference between MAC and Windows etc ?

bobh46320, 2017-01-19:
Whoops,  I didn’t know that I closed the ticket.  I replied with a stack dump - and close &comment - I guess I should have used Comment.
Bob

ntoll, 2017-01-19:
Hmm... could you try the latest Mu available from https://codewith.mu/ please..? There's absolutely no difference between the Windows, Mac and Linux versions - it's a Qt5 application written in Python 3.

carlosperate, 2017-01-19:
Also I am bit concerned that the crash deletes the file. If you have a chance to test MU dated 18/01/2017 again, could you create the script, save it on a known location and confirm the crash deletes the file contents? I assume from your previous message that the file is still present, but empty?

carlosperate, 2017-01-19:
Also, for both issue it would help if you could execute Mu from the terminal running: `./mu.app/Contents/MacOS/mu`

bobh46320, 2017-01-19:
Downloaded MU from https://codewith.mu/ <https://codewith.mu/>.  That was created on 6 Nov 2016 - MU still crashes out.  I am going to re-boot the MAC in case something is happening in the background.
Bob

bobh46320, 2017-01-19:
Running from the terminal, it works, displays ??

pc5:~ Bob$ cd documents
pc5:documents Bob$ cd microbit
pc5:microbit Bob$ cd mu
pc5:mu Bob$ ls
mu 4.app	mu-3.app	mu-5.app	mu.app
pc5:mu Bob$ ./mu.app/Contents/MacOS/mu
Logging to /Users/Bob/Library/Logs/mu/mu.log
pc5:mu Bob$ 

However, Now I can’t run MU from ‘MAC windows’.

I will have to re-boot,  but will wait until you answer this mail
Bob
> On 19 Jan 2017, at 14:24, Carlos <notifications@github.com> wrote:
> 
> ./mu.app/Contents/MacOS/mu

carlosperate, 2017-01-19:
> However, Now I can’t run MU from ‘MAC windows’.

I am not quite sure I understand what you meant. Is the problem that the program runs in the terminal but no GUI window appears? That's a bit odd, could you check the contents of ` /Users/Bob/Library/Logs/mu/mu.log`

bobh46320, 2017-01-19:
When I run MU from the terminal,  the GUI works normally, I can save the source code file, I can Flash the file to the MicroBit and the Microbit runs OK and displays the ?? for a non recognised character.  I have quit MY, and quit TERMINAl, BUT now,  I can’t run MU directly in MAC windows.  The MU icon comes up on the screen, then disappears. I have looked on the activity monitor,  but can’t see any reference to MU running. I suspect a Re-boot will sort it, but I will wait in case you want something else from me.

carlosperate, 2017-01-19:
No, thanks, I guess it's better to try again after a reboot. A bit weird the application doesn't open by double clicking the mu.app icon anymore. You could have a look at the log in case there is something there, but it's probably better to see if it works again after the reboot.

(@ntoll: This reminds me there is another PR I've been seating on for a while to print exceptions in the log and move it as "mu_crash.log" in the Mu application folder. I'll see if I can clean it up and prepare a PR this evening)

ntoll, 2017-01-19:
Hi,

There's a real danger that we may get our wires crossed. Bear with me while I try to unpick what's happening. ;-)

To be clear, the name of the application is "Mu", not "Mu5" - I assume (and perhaps you can confirm) that you may have downloaded Mu five times and this is an artefact of repeated downloads since OSX appends a number for each re-download.

I'm not clear what you mean by "MAC windows". I could imagine all sorts of things you could mean (you're running Mu in some sort of Windows based virtualisation [such as Parallels] on your Mac, or perhaps you mean you simply can't get the app to appear as a window on your desktop).

Apologies if this all sounds a bit silly, but I want to be absolutely clear what you mean.

I've used Mu (for instance the one found on the codewith.mu website) on my Mac with absolutely no problem. Many hundreds of other people use Mu on Macs without any problem every day without it crashing which suggests something local to you. I see you're on a recent version of OSX so I'm left wondering what other extenuating circumstances may make your machine problematic. Any ideas?

Perhaps, as you suggest, it's simply a case of "have you tried switching it off and on again?" -- i.e. see if a reboot helps.

Sorry I can't be more helpful. :-/

ntoll, 2017-01-19:
@carlosperate great idea. I wonder about a crash-report "wizard" built into the app. Might be overkill, but we'd be able to collect some very useful information.

bobh46320, 2017-01-19:
OOPS,  that’s screwed a few things up. I can’t run MU by clicking on the MU file.  The icon comes up on the screen, then closes down again without leaving the GUI screen there.  This happens with a couple of different versions of MU including the latest version dowloaded today. What did your code do to my MAC,  as a re-boot hasn’t helped ? Also I can’t find a log file anywhere.

pc5:~ Bob$ cd documents
pc5:documents Bob$ cd microbit
pc5:microbit Bob$ cd mu
pc5:mu Bob$ ls
mu 4.app	mu-3.app	mu-5.app	mu.app
pc5:mu Bob$ ./mu.app/Contents/MacOS/mu
Logging to /Users/Bob/Library/Logs/mu/mu.log
pc5:mu Bob$

bobh46320, 2017-01-19:
Found the log file:-
Last login: Thu Jan 19 15:07:48 on console
pc5:~ Bob$ cd /Users/Bob/Library/Logs/mu
pc5:mu Bob$ ls
mu.log
pc5:mu Bob$ more mu.log
root(run) INFO: Starting Mu 0.9.13
mu.logic(__init__) INFO: Setting up editor.
mu.logic(get_settings_path) INFO: Application directory: /Users/Bob/Documents/MicroBit/MU
mu.logic(get_settings_path) INFO: Application directory: /Users/Bob/Documents/MicroBit/MU
mu.logic(restore_session) INFO: Restoring session from: /Users/Bob/Library/Application Support/mu/settings.json
mu.logic(restore_session) DEBUG: {'paths': ['/Users/Bob/Documents/MicroBit/SOURCE/LCD temperture.py', '/Users/Bob/Documents/MicroBit/SOURCE/LCD OptoCounter.py', '/Users/Bob/Documents/MicroBit/SOURCE/LCD OptoCounter.py', '/Users/Bob/mu_code/thermometer.py', '/Users/Bob/mu_code/angle.py', '/Users/Bob/mu_code/Buzzer.py', '/Users/Bob/Documents/MicroBit/LCD MicroBit.py', '/Users/Bob/Downloads/degreetest3.py'], 'workspace': '/Users/Bob/mu_code', 'theme': 'day'}
pc5:mu Bob$

ukBaz, 2017-01-19:
Just a suggestion to help with the debugging and better visualize what steps are being taken. Would it help to capture a screen cast of what is being done. On a Mac this can be done with QuickTime.
QuickTime is free and comes pre-installed with OS X (10.6 or above). Launch Quick Time, go to File > New Screen Recording in the QuickTime menu, select your options and begin recording. To record all or part of your screen, just click and drag the area you wish to capture. You can even set it to record “blips” (or visual cues) for every mouse click.

bobh46320, 2017-01-19:
OOPS, file too big to send 20 mb,  I will try with Blips?

ukBaz, 2017-01-19:
Do you have a youtube account you could (temporarily?) upload the video to?

bobh46320, 2017-01-19:
can you handle split files.  Wait for all 3 files to arrive please.

ntoll, 2017-01-19:
Hmm... the log file suggests it's getting as far as trying to reload the files you had open when you last closed Mu. My guess is one of these files contains something that Mu doesn't like (it'd be interesting find discover what that file contains).

As a temporary measure, please delete the session file: `/Users/Bob/Library/Application Support/mu/settings.json` and try starting Mu again. It should open in an empty "vanilla" state. 

It'd be really helpful if you could try loading the files you had open in your last session to see if one of them breaks Mu ('/Users/Bob/Documents/MicroBit/SOURCE/LCD temperture.py', '/Users/Bob/Documents/MicroBit/SOURCE/LCD OptoCounter.py', '/Users/Bob/Documents/MicroBit/SOURCE/LCD OptoCounter.py', '/Users/Bob/mu_code/thermometer.py', '/Users/Bob/mu_code/angle.py', '/Users/Bob/mu_code/Buzzer.py', '/Users/Bob/Documents/MicroBit/LCD MicroBit.py', '/Users/Bob/Downloads/degreetest3.py').

If / once the gremlin file is identified, could you please attach it to this ticket?

Then, of course, re-delete the session file as per the instructions above to reset Mu.

Does this help?

bobh46320, 2017-01-19:
Grr,  5mb files fail, so delete any files received so far *.spt and *.jhiospt
No,  I don’t have youtube a/c either

carlosperate, 2017-01-19:
You could try something like http://transfer.sh

ntoll, 2017-01-19:
@bobh46320 I wouldn't worry too much about uploading a screencast of the problem. Try my suggestion from 9 minutes ago - that'll probably give us more useful information (although I'm off to a meeting now, so will follow up this evening).

bobh46320, 2017-01-19:
Well spotted!  I deleted the .py file successfully created by Mu run at the terminal command, and now all is OK.  BUT, We are now back where we started i.e..  If I run MU from the terminal,  it works OK and accepts “°” (the degree symbol). Then I must delete the saved file.  If I now run MU via MAC windows (Ie.  just clicking on the file), the GUI screen comes up,  I enter the code, save the code to a file and MU then quits (As it is trying to save the degree symbol). The source file is saved but is empty and I can re-launch MU,  but the source code file is empty.  Its a repeatable bug.
Bob

ntoll, 2017-01-19:
Aha! We're making progress (and quickly before my meeting starts).

I can't reproduce this bug. I *can* save the file. Can you paste the log file from Mu (like you did before from `/Users/Bob/Library/Logs/mu/mu.log`) immediately AFTER you've recreated this problem. I hope to be able to see a Python related error relating to your original bug. :-)

bobh46320, 2017-01-19:
degreetest2.py is a file which was created by MU (not using terminal) and is empty - MU then crashed out
degreeetest.py is a file created by MU at the terminal and has the 2 lines of code in it. - I have to delete this file before I can get MU to run outside of MAC terminal
Your help is appreciated and I hope you can resolve this problem for me and others.

bobh46320, 2017-01-19:
Log file :-
root(run) INFO: Starting Mu 0.9.13
mu.logic(__init__) INFO: Setting up editor.
mu.logic(get_settings_path) INFO: Application directory: /private/var/folders/77/bhmnp6s913g05t14lz64fty80000gn/T/AppTranslocation/04015358-AA65-4B35-BA3D-BB27AFE0A57C/d
mu.logic(get_settings_path) INFO: Application directory: /private/var/folders/77/bhmnp6s913g05t14lz64fty80000gn/T/AppTranslocation/04015358-AA65-4B35-BA3D-BB27AFE0A57C/d
mu.logic(restore_session) INFO: Restoring session from: /Users/Bob/Library/Application Support/mu/settings.json
mu.logic(restore_session) DEBUG: {'paths': ['/Users/Bob/Documents/MicroBit/SOURCE/LCD temperture.py', '/Users/Bob/Documents/MicroBit/SOURCE/LCD OptoCounter.py', '/Users/Bob/Documents/MicroBit/SOURCE/LCD OptoCounter.py', '/Users/Bob/mu_code/thermometer.py', '/Users/Bob/mu_code/angle.py', '/Users/Bob/mu_code/Buzzer.py', '/Users/Bob/Documents/MicroBit/LCD MicroBit.py'], 'theme': 'day', 'workspace': '/Users/Bob/mu_code'}
mu.logic(get_settings_path) INFO: Application directory: /private/var/folders/77/bhmnp6s913g05t14lz64fty80000gn/T/AppTranslocation/04015358-AA65-4B35-BA3D-BB27AFE0A57C/d
mu.interface(get_save_path) DEBUG: Getting save path: /Users/Bob/Downloads/degree3
mu.logic(save) INFO: Saving script to: /Users/Bob/Downloads/degree3.py

deshipu, 2017-01-19:
It seems that the files you are attaching to your e-mail do not appear on this ticket. Can you attach them using the web interface instead, by visiting this url? https://github.com/mu-editor/mu/issues/231

bobh46320, 2017-01-19:
Log file :-
root(run) INFO: Starting Mu 0.9.13
mu.logic(__init__) INFO: Setting up editor.
mu.logic(get_settings_path) INFO: Application directory: /private/var/folders/77/bhmnp6s913g05t14lz64fty80000gn/T/AppTranslocation/04015358-AA65-4B35-BA3D-BB27AFE0A57C/d
mu.logic(get_settings_path) INFO: Application directory: /private/var/folders/77/bhmnp6s913g05t14lz64fty80000gn/T/AppTranslocation/04015358-AA65-4B35-BA3D-BB27AFE0A57C/d
mu.logic(restore_session) INFO: Restoring session from: /Users/Bob/Library/Application Support/mu/settings.json
mu.logic(restore_session) DEBUG: {'paths': ['/Users/Bob/Documents/MicroBit/SOURCE/LCD temperture.py', '/Users/Bob/Documents/MicroBit/SOURCE/LCD OptoCounter.py', '/Users/Bob/Documents/MicroBit/SOURCE/LCD OptoCounter.py', '/Users/Bob/mu_code/thermometer.py', '/Users/Bob/mu_code/angle.py', '/Users/Bob/mu_code/Buzzer.py', '/Users/Bob/Documents/MicroBit/LCD MicroBit.py'], 'theme': 'day', 'workspace': '/Users/Bob/mu_code'}
mu.logic(get_settings_path) INFO: Application directory: /private/var/folders/77/bhmnp6s913g05t14lz64fty80000gn/T/AppTranslocation/04015358-AA65-4B35-BA3D-BB27AFE0A57C/d
mu.interface(get_save_path) DEBUG: Getting save path: /Users/Bob/Downloads/degree3
mu.logic(save) INFO: Saving script to: /Users/Bob/Downloads/degree3.py

The 'saved' degree3.py file is empty after MU crashes and it is not possible to attach it here (Because it is empty)

deshipu, 2017-01-19:
How about the file you get when you save in a Mu while it's running from a terminal?

bobh46320, 2017-01-19:
Strange I can't attach a .py file,  I will change the file type to .txt,  you will have to change it back to .py for me ?? I appreciate the name has changed,  but the contents are the same.
[degreetest3.py.txt](https://github.com/mu-editor/mu/files/717170/degreetest3.py.txt)

deshipu, 2017-01-19:
Thank you, that works.

deshipu, 2017-01-19:
So the characters giving this problem are "\xc2\xb0". When I open that file in Mu under Linux, I get this:
[image]

bobh46320, 2017-01-19:
Yep,  Decimal 176 graphical Degree symbol,  but If I open it as a TXT file in a bland editor,  it displays a degree symbol thus   °

inglesp, 2017-01-19:
I can reproduce this using mu-0.9.13.osx.

To summarise:

If I launch mu by double clicking on the mu app icon in Finder, mu crashes when I try to save a new file containing a degree symbol.

However, if I launch mu from the terminal, I am able to save a new file containing a degree symbol.

When mu crashes, I get the following in my mu log:

```
root(run) INFO: Starting Mu 0.9.13
mu.logic(__init__) INFO: Setting up editor.
mu.logic(get_settings_path) INFO: Application directory: /Users/inglesp
mu.logic(get_settings_path) INFO: Application directory: /Users/inglesp
mu.logic(restore_session) INFO: Restoring session from: /Users/inglesp/Library/Application Support/mu/settings.json
mu.logic(restore_session) DEBUG: {'theme': 'day', 'workspace': '/Users/inglesp/mu_code', 'paths': []}
mu.logic(get_settings_path) INFO: Application directory: /Users/inglesp
mu.interface(get_save_path) DEBUG: Getting save path: /Users/inglesp/mu_code/degree.py
mu.logic(save) INFO: Saving script to: /Users/inglesp/mu_code/degree.py
```

I also get a system dialogue box saying "mu quit unexpectedly", and gives me the option of sending a problem report to Apple.  The contents of the report is in [this gist](https://gist.github.com/inglesp/31fe76130a731dde5373900ba16177d8).

I'll try to dig into this further tonight.

carlosperate, 2017-01-19:
Since we are not forcing an encoding format for the files in Python I think it falls back to the system default, which is platform dependent and the reason we only see this in OS X. It is possible that the default encoding might also differ in OS X if running in a terminal vs the desktop GUI?

So, when running Mu from the terminal you are able to save the file correctly and it does **not** crash, is that correct?
Can you then open Mu from the terminal again and see the file displaying the file correctly?

carlosperate, 2017-01-19:
So this will probably take a couple of hours to build, but @inglesp if you are free before that maybe you could apply the patch and test if this makes a difference?: https://github.com/mu-editor/mu/commit/e6f637f867509464b7ba9eb814ef16ced043ef92

https://travis-ci.org/mu-editor/mu/jobs/193492599

bobh46320, 2017-01-19:
Correct,  If I run MU from terminal, it works fine. I can save the source, flash it, add more code to the source, save it and flash it.  Attached are 2 log files.  MUterminal is the log file when run from Terminal. MUwindow is the log file as it crashes when run from the GUI window.

[mu Windowcopy.log.txt](https://github.com/mu-editor/mu/files/717572/mu.Windowcopy.log.txt)
[mu Terminalcopy.log.txt](https://github.com/mu-editor/mu/files/717573/mu.Terminalcopy.log.txt)

ntoll, 2017-01-19:
I'm back from London after a bunch of meetings. Just wanted to say you folks rock! :-)

I'll follow up in the morning when I've caught up with some sleep..!

Gotta :heart: the Python community

carlosperate, 2017-01-19:
The test built I previously mentioned is done: https://s3-us-west-2.amazonaws.com/ardublockly-builds/microbit/osx/mu-2017-01-19_22_03_54.zip

inglesp, 2017-01-19:
@carlosperate That new version works for me.  Good stuff!

How can we avoid this happening again?  Your fix in e6f637f867509464b7ba9eb814ef16ced043ef92 addresses all existing cases of opening a file, but what happens when we introduce a new file and forget to set the encoding?

carlosperate, 2017-01-19:
Glad it works Peter! I can't access a mac today, so it was a very quick test (branch name...) to see if we could confirm if this was indeed the problem.

I must admit I am not quite sure what might be the best solution though. We could ensure Mu always works with UTF-8 internally, but I am not sure if there is a reliable way to detect the encoding of an unknown file created by a different editor.

Also, I haven't really had a chance to have a proper look at the dump, so I am not 100% sure where the application is crashing. Since you haven't seen any Python exception or trace printed in the console I would assume the problem happens internally in the Qscintilla or PyQt libraries, so maybe there could be an internal encoding conversion there that fails and we could sanitise beforehand within Mu? Could this be a problem only on the OS X PyQt/Qscintillar binaries? Or could it be present in all platforms, but only a problem with the default OS X encoding? We don't really know until we try to debug this on a system where the issue can be replicated.

ntoll, 2017-01-20:
@carlosperate great work!

Ugh... so I've spent much of the morning so far reading up about unicode, UTF-8 and Python's `open` builtin.

Regarding detection of encoding of a file created by a different editor: I suggest we only guarantee to read files created by Mu itself. I also found the follow docs very useful: http://python-notes.curiousefficiency.org/en/latest/python3/text_file_processing.html - they suggest several approaches and potential pitfalls.

I'm not sure I've made up my mind, but I like the idea of enforcing `utf-8` for Mu. It's explicit and we cover our backs with the guarantee/caveat mentioned above. Furthermore, we could handle the potential pitfalls outlined with this approach in the document referenced above - for example, falling back to `surrogateescape` as the type of unicode error handling used by Python.

Does this make sense? In any case, the document linked to above is readable and helped me clarify my thoughts - although I remain open minded to counter suggestions.

Thoughts..?

ntoll, 2017-01-20:
@inglesp regarding your question about introducing new code that opens a file, I guess the only thing we can do is abstract this away into functions that do the right thing. This should be documented somewhere - @tjguk and I discussed yesterday evening writing up such technical decisions in our documentation so new developers have something to refer to when they reach a "ZOMG WTF?" situation.

inglesp, 2017-01-20:
I wasn't necessarily talking about introducing new code that opens a file.  If we're working under the hypothesis that the difference in behaviour between launching the app from Finder and from the terminal is caused by having different default encodings, I'd hope it was possible to set the default encoding to always be UTF-8.  That way, we wouldn't have to specify this every time we open a file.  (I'm waving my hands a bit here: I'm not sure that "default encoding" is the right phrase for what we want.)

Unfortunately I haven't been able to build mu from source and have limited time to work on this today.

ntoll, 2017-01-20:
Yes, I see what you mean. I couldn't find a way to specify a default encoding either - but perhaps my Google-fu is problematic.

Like you I have limited time today.

inglesp, 2017-01-20:
According to the [docs on `open`](https://docs.python.org/3/library/functions.html#open), the default encoding is "whatever `locale.getpreferredencoding()` returns".

I have now managed to build mu from source and logging the return value of `getpreferredencoding()` shows that when the app is launched from terminal it is "UTF-8", and when launched from Finder it is "US-ASCII".

At the moment I have no idea whether it's possible to change what `getpreferredencoding()` returns.

I think there are three bugs here:

 * mu crashes when it tries to save a file containing unicode if it has been launched via OSX Finder (and possibly in other scenarios where `locale.getpreferredencoding()` is not "UTF-8").  This can be fixed by [always opening files with `encoding='utf-8'`](e6f637f867509464b7ba9eb814ef16ced043ef92), but it'd be nice to find a way to set the preferred encoding.
 * mu crashes when it tries to open a file with an unexpected encoding.  We should instead display a message explaining that the file cannot be opened.
 * mu crashes without a traceback.  A traceback in the logs would've saved a lot of toing and froing on this issue.  We should wrap all of `app.run()` in a `try: ... except: ...`, and log the traceback before quitting.

If I have time later I'll create three issues to track these three bugs, but shan't mind if somebody else does so first!

deshipu, 2017-01-20:
I think the correct good practice is to always handle unicode inside your application, but always read/write bytes, whether it's with files or with network. That means that a correctly written Python program will explicitly convert unicode to bytes and bytes to unicode on input/output operations, using an explicit encoding of author's choice.

inglesp, 2017-01-22:
The reason for the difference in behaviour between an app launched via the terminal and an app launched via Finder is that an app launched via terminal inherits environment variables from the terminal session.

This is likely to include `LANG`, which for me is set to `en_GB.UTF-8`.  `LANG` is not set when launching the app via Finder.

`LANG` is what Python ultimately uses to in `locale.getpreferredencoding()`.  As far as I can tell there is no way to change what `locale.getpreferredencoding()` returns once Python has started, which means @carlosperate's [quick-n-dirty fix](https://github.com/mu-editor/mu/commit/e6f637f867509464b7ba9eb814ef16ced043ef92) is the way to go.  I have opened #237 to track this.

I have also submitted #234 to fix the problem of mu crashing without a traceback, and opened #236 to track giving the user a sensible error message when mu tries to open a file with an unexpected encoding.

I'm going to close this issue now as I believe its constituent parts are tracked elsewhere.

closed 2017-01-22 by inglesp