mu-editor/mu #232: Zoom renders small characters unreadable

romilly, 2017-01-21:
If you zoom in a lot, text containing characters like ' become unreadable because of aliaising.

Maybe Zoom should increase the font size rather than magnifying the viewport. Of course, that would affect wrapping, but that might be preferable behaviour.

carlosperate, 2017-01-22:
We do increase the font size for the zooming functionality. I generally do not get any aliasing issues, but I haven't set the zoom very all that often. Which Operating System and version of Mu are you using? could you include as screenshot as well?

romilly, 2017-01-23:
I'm running raspbian 4.4.38-v7+ on a Pi 3.
mu is version 0.1 (from the raspberry pi repo)
Screenshot attached:

​

carlosperate, 2017-01-23:
The version for the pi at the moment is quite old (that will soon change though!) but even that version should have been using the same method of zooming. 

About the screenshot, attachments don't work via email, you'll have to add the image from the GitHub website (you can drag and drop an image file into the comment).

ntoll, 2018-06-04:
Hmm... not sure what we can do since we appear to be doing the right thing and Raspbian isn't (for reasons I can't fathom). Closing (sorry).

closed 2018-06-04 by ntoll