mu-editor/mu #233: Add additional night mode

obrasier, 2017-01-22:
I think the current dark mode could be improved by having a high-contrast mode for the vision impaired, as well as a night mode with pretty syntax highlighting. (e.g. Sublime Text)

See my picture for comments:
[image]

carlosperate, 2017-01-22:
Thanks for the feedback Owen, as discussed in the slack channel we can definitely rotate between themes on each click to the toolbar button.

Adding this should also clarify the difference between a dark theme and high contrast, so we should probably also do some work to use less shades of grey on dark contrast.

P.D. @ZanderBrown (writing this here as there is no privates in GH): are you aware of the [microbit slack channel](http://tech.microbit.org/get-involved/where-to-find/)?

ZanderBrown, 2017-01-22:
I had heard it mentioned

ntoll, 2018-06-04:
Done. Closing. See latest beta builds.

closed 2018-06-04 by ntoll