mu-editor/mu #238: Zooming should change the width of the line number column

carlosperate, 2017-01-22:
When the zoom is changed, the column on the left of the editor displays the line numbers stays the same width (the font size in that column does change as expected). Not much a problem when zooming out, but when the font size is large enough some of the characters are obscured due to the limited space:

[image]

Not quite sure how much control we've got over this one, but we should check if this width can be changed accordingly.

ntoll, 2018-06-04:
We don't have control except for the fixed width of the column. I guess saying "don't do that" is the best we can do. :-/

carlosperate, 2018-06-04:
Can't the width of the column not be changed after it has been set? If so the zoom level could change this value as well?

carlosperate, 2018-06-04:
Can't we change the margin values while it is running by using [setMarginsBackgroundColor](http://pyqt.sourceforge.net/Docs/QScintilla2/classQsciScintilla.html#a419ab8aed49ea1711ce4ffcf19146df1) and [setMarginWidth](http://pyqt.sourceforge.net/Docs/QScintilla2/classQsciScintilla.html#a1a8b798b6b36db2b5f60b0cdccba88a1)?

ZanderBrown, 2018-06-04:
From my experience no

carlosperate, 2018-06-04:
Was there any specific issue with it? A simple hack on the code seems to change the GUI as expected (I just picked a different predefined theme item to the margin background):

[image]

ntoll, 2018-06-04:
@carlosperate good stuff (on resizing the width). Please create a PR. ;-)

ntoll, 2018-06-05:
Care to link to your branch with the spike code / proof-of-concept..?

carlosperate, 2018-06-05:
Pushed the ugly hack for reference: 7df7af9123c763b88924d0153a27c0a9bf6370fd

ntoll, 2018-12-18:
This appears to have been fixed!

carlosperate, 2018-12-18:
Just tested the latest CI build from master and it looks more or less the same to me. There is a bit more margin, but still not dynamic width: 

[image]

ntoll, 2018-12-18:
Doh... sorry... my misunderstanding... will re-open.

dybber, 2020-12-10:
Here's an attempt at this:

```diff
diff --git a/mu/interface/editor.py b/mu/interface/editor.py
index ece9398..0fdc12a 100644
--- a/mu/interface/editor.py
+++ b/mu/interface/editor.py
@@ -283,6 +283,10 @@ class EditorPane(QsciScintilla):
             "xxl": 16,
             "xxxl": 48,
         }
+        # Make the margin to the left of line numbers follow
+        # zoom-level with a minimum of 50 pixels, and increasing
+        # slightly on each zoom level
+        self.setMarginWidth(0, 50 + (sizes[size]+10)/2)
         self.zoomTo(sizes[size])

     @property
```

[image]

PR #1311 (2021-02-19): Make margins around debugger-marker follow zoom-levels
Follow up on #1310 to fix #238
closed 2021-02-19 by ntoll