mu-editor/mu #239: How to use in Ubuntu 16.04 64bit???

KimHwaMok, 2017-01-23:
I tried "python3 run.py" but "ImportError: No module named 'pyQt5'"
so I installed "sudo apt-get install python-qt4"
but same happen again.

How to use in Ubuntu 16.04 64bit?

ntoll, 2017-01-23:
Hi,

Check out the developer instructions in the README. Essentially you need: python3-pyqt5, python3-pyqt5.qsci and python3-pyqt5.qtserialport

closed 2017-01-23 by ntoll

KimHwaMok, 2017-01-23:
@ntoll 
hi, thank you so much for reply.
I have completed the installation for the mentioned things to solve all dependency problems.
error code is "No module named 'pyQt5.Qsci'"
so I tried to "apt-get install python3-pyqt5.qsci" but same happen.

ZanderBrown, 2017-01-23:
have you tryed `sudo pip install requirements.txt` followed by `sudo python3 setup.py install`

carlosperate, 2017-01-23:
Are you sure you are using Python 3? So running `python3 run.py` instead of `python run.py`?

Could you execute the following lines individually in the command line:

```
python3 -c "import PyQt5;print('PyQt5 installed')"
```

```
python3 -c "import PyQt5.Qsci;print('Qsci installed')"
```

```
python3 -c "import PyQt5.QtSerialPort;print('QtSerialPort installed')"
```