mu-editor/mu #241: Hi, I have a few questions!

KimHwaMok, 2017-01-24:
I am a student studying a micro:bit. and I have a few questions.
1. I do not know what the features of Files. This button does not work when micro:bit are connected to the computer.
2.  I do not know what the features of Repl. This button does not work when micro:bit are connected to the computer.
3.  I modified Micro Python to support "scroll" functions in other languages. (ex, Korean) and I then created a new firmware through the "make all" command. and reflash my firmware through the "cp build/bbc-microbit-classic-gcc-nosd/source/microbit-micropython.hex /media/pllab/MICROBIT/" command.
but If i run in mu-editor, the result is same. nothing change. What's wrong with my way?

carlosperate, 2017-01-24:
Hi,

For general questions the best place to ask is probably the microbit python email list: https://mail.python.org/mailman/listinfo/microbit

1. You might find these links useful
http://microbit-micropython.readthedocs.io/en/latest/tutorials/storage.html
http://microbit-playground.co.uk/howto/add-python-module-microbit-micropython
2. Which operating system are you using? If Windows, have you installed the drivers?
http://microbit-micropython.readthedocs.io/en/latest/devguide/repl.html
3. Mu packs a version of MicroPython inside, so every time you flash your code it erases all of flash and loads micropython + your script. If you want to load a custom version of MicroPython you'll use to use a toold like uFlash: http://uflash.readthedocs.io/en/latest/

carlosperate, 2017-09-27:
I hope I was able to answer all your questions @KimHwaMok, since it's been quite a few months since this issue was opened I will close it as resolved. But, please feel free to reopen it if you have any further questions.

closed 2017-09-27 by carlosperate