mu-editor/mu #242: In windows if you open the file's panel when an error is still scrolling on the microbit it locks until the message has completed and then is blank

gingemonster, 2017-01-28:
Here is how I was replicating this:

1) Write main.py that imports a class from another file
2) Open the other file and main.py in mu
3) Flash main.py to microbit, it then complains it cannot find the class from the other file as its not been added via files panel yet
4) wait until error message finishes on microbit before opening files panel otherwise it hangs and when it loads is empty and if you click files or repl now it complains there is no microbit connected
5) open files menu and copy other file with class in to microbit
6) reset microbit and all is now working

but you can just create a main.py with an error.

carlosperate, 2017-11-03:
I hope you don't mind I edited the original title, now it's a bit shorter and would include other issues triggered on the same steps, like https://github.com/mu-editor/mu/issues/252.

Also, slightly related: https://github.com/mu-editor/mu/issues/291

ntoll, 2018-06-04:
This has been fixed (you get a helpful error message). Closing.

closed 2018-06-04 by ntoll