mu-editor/mu #244: Investigate Travis segmentation fault during test

carlosperate, 2017-02-03:
When py.test is run under the Travis linux image it always fails during collection: https://travis-ci.org/mu-editor/mu/jobs/187949922#L380

```
$ py.test --cov-config .coveragerc --cov-report term-missing --cov=mu tests/

============================= test session starts ==============================

platform linux -- Python 3.5.2, pytest-3.0.5, py-1.4.32, pluggy-0.4.0
rootdir: /home/travis/build/mu-editor/mu, inifile: 
plugins: cov-2.4.0
collecting 2 items/home/travis/build.sh: line 57:  3327 Aborted                 (core dumped) py.test --cov-config .coveragerc --cov-report term-missing --cov=mu tests/

The command "py.test --cov-config .coveragerc --cov-report term-missing --cov=mu tests/" exited with 134.
```

Under OS X it seems to fail occasionally with the following issue, but restarting the build usually solves it:

```
py.test --cov-config .coveragerc --cov-report term-missing --cov=mu tests/

============================= test session starts ==============================
platform darwin -- Python 3.5.2, pytest-3.0.6, py-1.4.32, pluggy-0.4.0
rootdir: /Users/travis/build/mu-editor/mu, inifile: 
plugins: cov-2.4.0
collected 188 items 

tests/test_app.py ...
tests/test_interface.py ..........make: *** [coverage] Segmentation fault: 11

The command "if [ "$TRAVIS_OS_NAME" = "osx" ]; then make check; fi" exited with 2.
```
This issue has been present for a long time and has seen more often lately: https://github.com/mu-editor/mu/pull/243#issuecomment-277130113, https://github.com/mu-editor/mu/pull/156#issuecomment-248279367, https://github.com/mu-editor/mu/pull/103#issuecomment-211288636, https://github.com/mu-editor/mu/pull/230#issuecomment-273771037
The last time in https://github.com/mu-editor/mu/pull/243, it failed 4 times in a row, so this is something that we should investigate.

carlosperate, 2017-02-03:
The Travis job from #243 merged into master is exhibiting the same issue: 
https://travis-ci.org/mu-editor/mu/jobs/197946892#L337

ntoll, 2017-02-03:
This completely not my area of expertise, but a quick Google for "Segmentation fault: 11" leads me to this: http://stackoverflow.com/questions/12762944/segmentation-fault-11 (the accepted response suggests the seg-fault is due to lack of memory). 

Given that it's a flapping problem, i.e. we can re-run the job and it'll sometimes work and sometimes won't (it appears that the behaviour is non-deterministic), I wonder if this is an infrastructure problem? For example, perhaps sometimes the job is running on a VM with too little memory and other times it has enough. Is there any way to check? Is there any way to say, "we need a minimum of Xmb" for a job?

Thoughts..?

Like I said, this is completely NOT my area of expertise so take anything I suggest with a pinch of salt. :smiley:

carlosperate, 2017-02-03:
That could quite possible be the issue with the OS X image.
As far as I know there I no way to specify more resources (https://docs.travis-ci.com/user/ci-environment/#Virtualization-environments), but maybe if we can get a a more verbose output we can get a better idea during which test it falls, and maybe that could give us a hint as what could be happening.

We could also give this a try: https://github.com/abalkin/pytest-leaks or https://nvbn.github.io/2017/02/02/pytest-leaking/

carlosperate, 2017-02-08:
Just to post a quick update before I completely forget about it. Last weekend I repeated a travis build of the [travis_osx_seg_fault branch](https://github.com/mu-editor/mu/tree/travis_osx_seg_fault) and out of 20 jobs, 4 of them failed:

[log1.txt](https://github.com/mu-editor/mu/files/761262/log1.txt)
[log2.txt](https://github.com/mu-editor/mu/files/761259/log2.txt)
[log3.txt](https://github.com/mu-editor/mu/files/761260/log3.txt)
[log4.txt](https://github.com/mu-editor/mu/files/761261/log4.txt)

Two of them on the same test, but not the others. Even thought this is a very small sample, it doesn't seem to be cause by one specific test, so likely to be an issue with available resources as pointed out by @ntoll . Technically the VM should be getting 4GBs of RAM, so it'd be interesting if we could add pytest hooks to monitor memory resources between tests.

carlosperate, 2017-05-03:
As a quick note, apart from segmentation faults we have also encountered a `[coverage] Abort trap: 6` in https://travis-ci.org/mu-editor/mu/jobs/228044691

carlosperate, 2017-06-21:
We are getting these quite consistently now (3 errors in a row each):

https://travis-ci.org/mu-editor/mu/jobs/245462227 : [log.txt](https://github.com/mu-editor/mu/files/1093232/log.txt)
```
tests/test_interface.py ............make: *** [coverage] Abort trap: 6
```

https://travis-ci.org/mu-editor/mu/jobs/245475916 : [log.txt](https://github.com/mu-editor/mu/files/1093234/log.txt)
```
tests/test_interface.py ..................make: *** [coverage] Segmentation fault: 11
```

closed 2018-03-13 by ntoll