mu-editor/mu #247: How to run the modified version in linux on Windows?

KimHwaMok, 2017-02-08:
Hi, I have a question.
I have modified mu-editor in linux. and want to run in Windows.
How to run the modified version in linux on Windows?

tjguk, 2017-02-08:
The answer depends on how much you already know about setting things up 
on Windows. If you're already a competent Python-on-Windows developer 
then it's pretty much the same as you'd do for Linux:

(Assuming Python 3.5)

Install virtualenv
Create virtualenv
Activate virtualenv
git clone your repo
cd your repo
pip install -r requirements
pip install PyQt5 QScintilla

Then off you go.

If you're really asking "How do I run Python on Windows in general?" 
then feel free to come back. We definitely want to work up our docs so 
any developer can pick them up and use them. I don't know just how far 
back we need to go ("Go to python.org/downloads; Press [Save]; 
Double-click on the .exe file...") but at least we'll try to understand 
what would help.

TJG

carlosperate, 2017-09-27:
Hi @KimHwaMok, hope you were able to build your Mu version on Windows with @tjguk instructions (thanks Tim!). 
Since it's been several months since the original question, I will close this issue as resolved. Please do open it again if you need further assistance.

closed 2017-09-27 by carlosperate