mu-editor/mu #248: Black screen of death when using Mu on a Macbook air

whaleygeek, 2017-02-12:
Sometimes when I plug my micro:bit into my Macbook air 11" I get a black screen of death. It only does this while Mu is running. I've often seen it when plugging in the micro:bit while Mu is running, and also I have often seen it when using Mu to stream serial data from print statements via the REPL.

Sun Feb 12 21:22:32 2017

*** Panic Report ***
panic(cpu 0 caller 0xffffff800fe16df2): Kernel trap at 0x0000000000000000, type 14=page fault, registers:
CR0: 0x000000008001003b, CR2: 0x0000000000000000, CR3: 0x0000000012b22000, CR4: 0x00000000001627e0
RAX: 0xffffff801b0c0160, RBX: 0xffffff807a96b000, RCX: 0x0000000000000082, RDX: 0xffffff80104d73e0
RSP: 0xffffff80a8b33f38, RBP: 0xffffff80a8b33f50, RSI: 0xffffff801b3c08e0, RDI: 0xffffff801ab59c00
R8:  0xffffff7f907ca385, R9:  0xffffff80104d6a90, R10: 0xffffff8010354b30, R11: 0x000076fcfe55377d
R12: 0x0000000008000040, R13: 0xffffff8020b88e00, R14: 0x000000000e000040, R15: 0x000000000e000040
RFL: 0x0000000000010286, RIP: 0x0000000000000000, CS:  0x0000000000000008, SS:  0x0000000000000000
Fault CR2: 0x0000000000000000, Error code: 0x0000000000000010, Fault CPU: 0x0

Backtrace (CPU 0), Frame : Return Address
0xffffff80a8b33be0 : 0xffffff800fd2ad21 
0xffffff80a8b33c60 : 0xffffff800fe16df2 
0xffffff80a8b33e20 : 0xffffff800fe33ca3 
0xffffff80a8b33e40 : 0x0 
0xffffff80a8b33f50 : 0xffffff7f90bb4fe1 
0xffffff80a8b33fb0 : 0xffffff800fe116e7 
      Kernel Extensions in backtrace:
         com.apple.iokit.IOSerialFamily(11.0)[0453BE0D-DB4F-3A51-9CFC-9A93615CCF3E]@0xffffff7f90bb0000->0xffffff7f90bbdfff

Looks to me like the serial driver is calling address zero (look at the Kernel Trap At address in the first line of the crash report).

OSX version: OSX Yosemite Version 10.10.5
Macbook Air (11-inch, early 2014)
Processor: 1.4GHz Intel Core i5

whaleygeek, 2017-02-12:
It's ok, I've proved this is not Mu, it's something Mac related. I have seen this same error happen with a pyserial script now, and also a CoolTerm session. Closing, as clearly there is nothing Mu can do if the OS crashes.

closed 2017-02-12 by whaleygeek

ntoll, 2017-02-12:
Dammit... I was just in the middle of writing a detailed answer about checking it's not an OS related issue, then saw your replies. Glad it isn't Mu. :-)